Bodmer/TFT_eSPI #391: ST7789 240x240 for esp8266

lucasromeiro, 2019-07-08:
Hello, I'm using a NodeMcu with a ST7789 240x240 display.
But I did not find instructions on how to make the physical links in the documentation.
Also I did not find any examples of ST7789 with NodeMcu (esp8266).
Can you instruct me how to make the connections and where can I find an example to test the operation of the LCD?

Thank you

Bodmer, 2019-07-08:
There are a few ST7789 displays availble, provide a web link to your exact display so I can see the pin names of list all the pin names exactly as written on the circuit board.

lucasromeiro, 2019-07-08:
Hello, this is the link I bought the LCD.
How can I identify the difference that you speak?
I want to use the minimum number of pins to control it, what is the minimum possible?
Thanks for the quick help! You're great!

https://pt.aliexpress.com/item/32880846744.html?spm=a2g0s.9042311.0.0.3c9fb90atJKLJF

Bodmer, 2019-07-09:
Suggested connections are:

 Display BLK - no connection unless you want to control the backlight on/off
 Display DC to NodeMCU pin D3
 Display RES to NodeMCU pin D4
 Display SDA to NodeMCU pin D7
 Display SCL to NodeMCU pin D5
 Display VCC to NodeMCU 3.3V
 Display GND to NodeMCU pin GND (0V)

There are no examples that are specific to 240x240 displays. 320x240 examples may work but you might the screen may show garbage due to the off screen pixels getting wrapped back to the display. 160x128 examples will work but will not use all the screen. You can adapt the examples to suit your screen or just use them as a reference to the different functions.

Bodmer, 2019-07-09:
Note that D6 is part of the SPI port and thus behaves as a MISO signal and cannot be used for another function.

lucasromeiro, 2019-07-09:
Thanks for the answer!
Can I use soft spi?
I can use fewer pins because I need them for other functions.

Bodmer, 2019-07-09:
The TFT_eSPI library does not support soft SPI. The library does support overlap mode where the display shares the FLASH pins, then only 2 or 3 of the GPIO pins are needed but unfortunatley you need a display with a chip select line to use overlap mode.

liket, 2019-07-10:
I'm trying to run this exact type of screen as well.
Bodmer, have you tested your library on the ESP8266 recently?
Your library works *perfectly* with ST7789 connected to the ESP32, and speed is **hugely impressive**!
But, on an ESP8266 (Wemos D1 R1) with the pin assignments exactly as the example in Setup24_ST7789 (which matches what you wrote above), I get _no_ screen output -- backlight only. The serial console shows that the benchmark is completing, it's just that nothing is displayed.

However, the Arduino_ST7789 library _does_ give me screen output on the same hardware, but of course at snail speed (almost a full second to fill the screen).

So I wonder, could it be the case that a bug has snuck into the library since you last tested on an ESP8266, or am I missing something?

Of course I can also use the ESP32, but I have drawers full of ESP8266 boards so it's nice to be able to use them for something too. :)

Bodmer, 2019-07-11:
@liktek
Some users have had issues if they link the display RES (Reset) line to the ESP8266 power on reset line. If your setup is like this then try using a dedicated reset pin. The issue when using the power on reset pin appears to be that the ESP8266 SPI lines go through a period of hiatus at power up while the software boots and executes the setup() routines which set the logic line states, this logic line noise puts the TFT into a state where it ignores subsequent commands.  This situation arises because some displays do not bring the chip select (CS) line to a pin and thus the display accepts the hiatus period garbage commands while the software is booting. It is not clear why the Arduino_ST7789 library does not suffer in this scenario.

I will create a hardware setup and run some test over the coming weekend. In the meantime can you tell me exactly what pin connections you have used so I can duplicate them.

liket, 2019-07-11:
Thank you, Bodmer!
Interesting theory. I attempted to test it by adding the following code to the setup() function before tft.init():

```
  delay(1000);

  pinMode(TFT_RST,OUTPUT);
  digitalWrite(TFT_RST,false);
  delay(2000);
  digitalWrite(TFT_RST,true);
  delay(500);

  tft.init();
```

I ran that and I saw the *backlight* turn off for two seconds. That made me suspicious, because I happen to know that you can't control the backlight serially -- if you could, there wouldn't be any need for the BLK pin!

So I kept digging.. and dug some more.. and finally wrote an app where I could individually twiddle each GPIO pin of the board.

Turns out **the problem is actually in your pin assignments** in `user_setup_select.h`.

```
// These are the pins for all ESP8266 boards
//      Name   GPIO    Function
#define PIN_D0  16  // WAKE
#define PIN_D1   5  // User purpose
#define PIN_D2   4  // User purpose
#define PIN_D3   0  // Low on boot means enter FLASH mode
```

That assumption is incorrect... they are apparently NOT for all ESP8266 boards.

For this particular experiment, I was using THIS board:

[image]

I have DC connected to the pin silkscreened D3.. and RES connected to the pin silkscreened D2.

With my test app, I found out that those pins are actually GPIO5 and GPIO16!

So, I entered the correct pin numbers directly into the #define's in Setup24_ST7789.h so that they don't use your PIN_Dxx definitions:

```
#define TFT_CS   -1      // Define as not used
#define TFT_DC   5  // Data Command control pin
#define TFT_RST  16  // Reset pin (could connect to NodeMCU RST, see next line) //LC
```

..and after that it started working instantly!
So, THANK YOU -- I never would have discovered this if you hadn't mentioned about the reset pin. That wasn't it, but it pointed me in the direction of the real culprit.

HOLY CRAP your library is fast. Wow.
Could you tell me please -- how are you doing that??? :)
Or rather, what's wrong with every other library?
When I first noticed how slow other libraries were, I skimmed through the documentation to see if there were any accelerated commands, like filling the screen, and I couldn't find a single one! So I accepted the fact that the display was slow as molasses, it was still okay for displaying a static number even if it looks kinda stupid to take half a second to draw the screen.

The Arduino_ST7789 appears to be setting a high SPI baud rate as well, and I tried changing it. Setting the baud rate to 115,200 made it slower (and also made the WDT reset the board), but setting it to 27,000,000 made it no faster.

On the other hand, setting the baudrate to 921,600 with YOUR library makes it behave much like Arduino_ST77898 (until it crashes because of the WDT).

So anyway.
Might I recommend NOT having your own PIN_DXX definitions and instead using the DXX constants defined in the platform layer?

Because, I just checked mine ), and they

Bodmer, 2019-07-13:
You have a [rev. 1 board (R1)](https://forum.arduino.cc/index.php?topic=545113.msg4017789#msg4017789), so it looks like Wemos created their own port allocations which differed from the Arduino tools. I assume they realised their mistake because they corrected this in the R2 board [(see here](https://wiki.wemos.cc/products:d1:d1)). Watch out for this if you buy a newer corrected board.

closed 2019-07-15 by Bodmer