Bodmer/TFT_eSPI #392: Always show text in the middle

quorle, 2019-07-11:
Hello, I have another question. Is it always possible to display a text in the middle? The text, however, varies in length. Maybe this is possible

Bodmer, 2019-07-13:
Yes, use setTextDatum(), see the[ TFT_String_Align example](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/320%20x%20240/TFT_String_Align/TFT_String_Align.ino).

quorle, 2019-07-13:
Thanks for the quick answer, unfortunately that does not work for me. the code snippet looks like this:

```
if (Interpret_Aktuell_last != Interpret_Aktuell) {         // Vergleich der Aktuellen Titel (Alt) mit dem neuen Titel auf Veränderung. Erst bei Veränderung wird der neue
    Interpret_Aktuell_last = Interpret_Aktuell;            // neue Wert auf das Display geschrieben
    tft.setCursor(5, 85 ,4);                       // Cursor setzen für Anzeige aktuelle Titel ( Erste Zahl und Zweite Zahl = X (Breit) und Y (Höhe) Position; Dritte Zahl = Schriftgröße )

    tft.fillRect( 0, 85, 320, 25, TFT_BLACK );     // Aktuelle Bildpunkte bei neuem Wert auf Schwarz setzen, um kein Überschneiden des alten und neuen Wertes zu bekommen
    tft.setTextDatum(MC_DATUM);                                               // Bedeutung der Zahlen: erste Zahl = Displayposition Breite / Zweite Zahl = Dislpayposition Höhe / Dritte Zahl = Länge / Vierte Zahl = Breite
    tft.print(Interpret_Aktuell);                      // Titel aus Variable auf das Display schreiben
  }
```

Bodmer, 2019-07-13:
You need to use the library functions to use the datum as per example:
```
  int16_t  drawNumber(long long_num, int32_t poX, int32_t poY, uint8_t font),
           drawNumber(long long_num, int32_t poX, int32_t poY),
           drawFloat(float floatNumber, uint8_t decimal, int32_t poX, int32_t poY, uint8_t font),
           drawFloat(float floatNumber, uint8_t decimal, int32_t poX, int32_t poY),

           // Handle char arrays
           drawString(const char *string, int32_t poX, int32_t poY, uint8_t font),
           drawString(const char *string, int32_t poX, int32_t poY),
           drawCentreString(const char *string, int32_t dX, int32_t poY, uint8_t font), // Deprecated, use setTextDatum() and drawString()
           drawRightString(const char *string, int32_t dX, int32_t poY, uint8_t font),  // Deprecated, use setTextDatum() and drawString()

```

Bodmer, 2019-07-13:
Also the library provides a function to blank out old text providing you use a backgorund colour:

setTextPadding(uint16_t x_width);

See [example](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/320%20x%20240/TFT_Custom_Fonts/TFT_Custom_Fonts.ino#L128-L145).

quorle, 2019-07-13:
Ok, unfortunately I do not understand exactly what you mean by that. Unfortunately I can not find out how you did it so easy in your example to place the text always in the middle. Have already searched in the user_config.h, but there is also no entry that you can comment out. I always just overwrite individual lines, where practically a new text is written

Bodmer, 2019-07-13:
Assuming you are printing a string (or char array):
```
if (Interpret_Aktuell_last != Interpret_Aktuell) {         // Vergleich der Aktuellen Titel (Alt) mit dem neuen Titel auf Veränderung. Erst bei Veränderung wird der neue
    Interpret_Aktuell_last = Interpret_Aktuell;            // neue Wert auf das Display geschrieben
    tft.setCursor(5, 85 ,4);                       // Cursor setzen für Anzeige aktuelle Titel ( Erste Zahl und Zweite Zahl = X (Breit) und Y (Höhe) Position; Dritte Zahl = Schriftgröße )

    tft.fillRect( 0, 85, 320, 25, TFT_BLACK );     // Aktuelle Bildpunkte bei neuem Wert auf Schwarz setzen, um kein Überschneiden des alten und neuen Wertes zu bekommen
    tft.setTextDatum(TC_DATUM);                                               // Bedeutung der Zahlen: erste Zahl = Displayposition Breite / Zweite Zahl = Dislpayposition Höhe / Dritte Zahl = Länge / Vierte Zahl = Breite
    tft.drawString(Interpret_Aktuell, 160, 85);                      // Titel aus Variable auf das Display schreiben
  }
```

closed 2019-07-15 by Bodmer