Bodmer/TFT_eSPI #393: Sorry not a real issue but How could i ADD MCP_23S17 for parallel interface

arnold-b, 2019-07-14:
Hello, 

i have a few ILI9488 TFT but without SPI breakout. I would like to use it with mcp23S17. 
Could you give me some hint, how to do this ?

thanks

Bodmer, 2019-07-15:
The library uses a mix of SPI library calls and direct ESP32 register access, this complicates the code adaption. One way to identify the code that would need to be changed is to look for the sections that are conditionally compiled for the RPI board, I.e. look for RPI_ILI9486_DRIVER conditional statements.

Make sure you fully understand what each affected code section does before making changes. It is probably best to write independant test software to communicate with the TFT via the SPI to parallel chip first.

The way I would adapt the TFT_eSPI code is to first change it to make all communication via the SPI library, then rewrite the SPI library to make the use of the mcp23s17 transparent to the graphics library.

closed 2019-07-15 by Bodmer

arnold-b, 2019-07-15:
Thanks for advice