Bodmer/TFT_eSPI #395: ILI9488 touch inversed

elgerg, 2019-07-16:
Hi Bodmer,

Ive got a ILI9488 touch screen:

https://www.ebay.co.uk/itm/3-5inch-SPI-TFT-LCD-Serial-Module-Display-Screen-With-Touch-Panel-Driver-Hot/273830124378?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

Its not a 9486 but a 9488. It seems that the xy cords are inversed IE top left is 
y:471 x:310
bottom right 
x: 0     y: 0     z: 1397 

This is on   tft.setRotation(3);
Bu the same (opposite) happens on rotation 1.

Is there a variable i am missing to reverse the touch to screen co ords?

Thanks
Alex

Bodmer, 2019-07-16:
You need to calibrate the screen in the rotation you intend to use.

Have a look at [this](https://github.com/achillhasler/TFT_eTouch) library.

Bodmer, 2020-11-03:
Unable to reproduce the problem.

closed 2020-11-03 by Bodmer