Bodmer/TFT_eSPI #396: can't get display to work

mark-hahn, 2019-07-17:
<Note: I accidentally posted this in TFT_eFEX.  I closed it there.>

I have a "TFT LCD Screen Display 1.3inch TFT LCD Module, 240240 IPS 65K Full Color 3.3V with SPI Interface ST7789 IC Driver" module. I wanted to post an image but I can't figure out how to do that here.

I have a custom board with an esp-12. Everything compiles and loads fine. I'm trying this code ...

#include <Arduino.h>
#include <SPI.h>
#include <TFT_eSPI.h> 
TFT_eSPI tft = TFT_eSPI();
void setup() {
  tft.init();
  tft.fillScreen(TFT_RED);
}
void loop() {
  yield();
}
My setup is ...

TFT_eSPI ver = 1.4.5
Processor    = ESP8266
Frequency    = 80 MHz
Voltage      = 3.78 V
Transactions = No 
Interface    = SPI 
SPI overlap  = No 

Display driver = 7789
Display width  = 240 
Display height = 240 

R0 y offset = 80 
TFT_DC   = D8 (GPIO 15)
TFT_RST  = D3 (GPIO 0)

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled
Display SPI frequency = 10.0 MHz 
I know the setup is good because I see the pins working on the module. I hooked a scope up to all the active pins on my module: clock, data, reset, and cmd-data. Everything looks good. I even went through all the commands sent: setup, row and column addresses, and then the write. I can even see the red values in the data. The backlight is on but the screen is blank. I have tried two display modules.

Any ideas on what I should try next? Does anyone know a better forum to post this?

Bodmer, 2019-07-17:
You can drag and drop a jpeg image to the text window on a PC, eg just dragged and dropped this image:
[image]

List all connections between the EPS8266 and TFT, giving the GPIO and Arduino pin numbers.

Provide web links to the boards you have used.

mark-hahn, 2019-07-17:
Thanks for the reply.

Pins (numbers below are on the module shown at the bottom of the post) ...
```
1 GND
2 VCC (3.3V)
3 SCLK   <-->  GPIO14
4 MOSI   <-->  GPIO13
5 RESET   <-->  GPIO0  with 4.7k pullup and schottky diode --> DTR ("ck" reset method)
6 CTLDATA <--> GPIO15 with 10k pulldown
7 BACKLIGHT  <-->  VCC (3.3V)
```

As I said I am using a custom board ..

[image]

Display module is from is from https://www.amazon.com/gp/product/B07P9X3L7M/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

[image]

mark-hahn, 2019-07-18:
FYI: I'm going to try the adafruit gfx library as a test of my hardware.  I have no idea where to start so wish me luck.

mark-hahn, 2019-07-18:
I'm getting the exact same results with the adafruit library.  I checked the pins and read the commands with the scope as I did before.  I got the same blank screen.  I guess it's possible both displays I bought were bad, but I'm really reluctant to order more.  I guess I'll have to.  If that doesn't work I'll switch to an I2C display which would be a bummer.

I'm still looking for suggestions though.

Bodmer, 2019-07-18:
I think it is unlikely both displays are bad. I would try using a cheap NodeMCU board with the display as that is a known working setup.

Check your power supply is adequate and giving 3.3V under load.  The sharing of DTR and reset to the TFT may give issues since DTR may conflict with the ESP8266 output high level.

Try using the default pin connections.

mark-hahn, 2019-07-18:
>  I would try using a cheap NodeMCU board with the display as that is a known working setup.

Good idea.  I have some of those.  It would verify the display is good.  I can also compare scope signals.

> Check your power supply

Did that.

> DTR may conflict with the ESP8266 output high level.

DTR is always high after boot.  Verified with scope.  And problem exists with no uart connection (boot from flash).

> Try using the default pin connections.

I'll do that on the nodemcu.

Thanks very much for your expertise.   I'm desperate to get a product finished.

Bodmer, 2019-07-19:
I also suggest loading the latest version of the TFT_eSPI library. I will setup an identical NodeMCU and ST7789 display this weekend so I can check the setup still works. Some users have had issues with these displays so the are actually two #defines that can be used to set the driver type. would be worth trying:
#define ST7789_2_DRIVER
instead of:
#define ST7789_DRIVER

The former is a basic display initialisation I created, the latter is an enhanced display initialisation provided by a library user.

mark-hahn, 2019-07-21:
I tested with a wemos d1 mini and it worked.  I also tested a ST7735 on a 128x160 board.  It also worked. Both displays I tested were brand new and arrived today.

So now I'm going to set my custom board back up and try again.  I'm a little bit afraid that whatever the problem is will harm the displays.  I'll triple-check the power.

mark-hahn, 2019-07-22:
Everything works and I didn't change anything on my board.  Apparently both displays were bad.  I might have done something in my early testing to blow them up.  Not unusual.

Sorry to bother you.  Thanks for helping so much, way beyond the call of duty.

closed 2019-07-22 by mark-hahn

Bodmer, 2019-07-24:
Thanks for letting me know. Hope all goes well now.