Bodmer/TFT_eSPI #397: Question: Overlapping sprites

pfeerick, 2019-07-19:
I'm loving this library after having used the Adafruit libraries, and am slowly working on improving display performance by using features such as sprites, and all the helper functions you've crafted. 

However, I'm running into an issue, and want to know if I'm going mad, have written some truely horrendous code (WIP!!) or an unreported bug... namely, I was using a sprite to show text and a progress bar whilst in the setup phase of the program, destroyed it at the end, and then use a couple other sprites which are created and destroyed as used. However, it seems that two sprites which overlap the earlier progress bar sprite aren't displayed, even though that sprite was destroyed?

I'll mock up some MRE code to make sure I'm not going nuts, but was wondering if it was something that was possible?

Bodmer, 2019-07-20:
You should be able to do what you describe. You will need to provide some example code if you need help.

Are you using ESP32 or ESP8266?

pfeerick, 2019-07-21:
ESP8266

Went through the code, and this became the bare guts that only needs the TFT_eSPI 'n co libraries... and can't reproduce it now... so I must have screwed up somewhere. Back to the drawing board. 

Another question... since I'm using the TFT_eFEX library for the progress board... can I get it to draw the progress bar on the sprite? 

btw, I'm using platformIO, but this should compile on the Arduino IDE with no or minimal changes.

main.cpp
```
/* LCD Connections:
 * VCC -> VCC
 * GND -> GND
 * CS  -> D8
 * RST -> VCC
 * D/C -> D3
 * SDI -> D7
 * SCK -> D5
 * LED -> 47R -> VCC
 * SDO -< N.C.
 */
#include <Arduino.h>

#include "Free_Fonts.h" // Include the header file attached to this sketch
#include <SPI.h>
#include <TFT_eSPI.h>
#include <TFT_eFEX.h>

// function prototypes
void drawTemperature(String name, float temp, float temp_min, float temp_max, int poX, int poY);
void drawProgress(uint8_t percentage, String text);

// variables and objects
TFT_eSPI tft = TFT_eSPI();
TFT_eFEX fex = TFT_eFEX(&tft);
TFT_eSprite progress_spr = TFT_eSprite(&tft); 
TFT_eSprite temp_spr = TFT_eSprite(&tft);

boolean updateDisplay = false;
boolean firstRun = true;
float temp1 = 0.0;
float temp2 = 5.0;
unsigned long lastUpdate = 0;

void setup()
{
  Serial.begin(115200); //default baud after boot
  Serial.println("");

  tft.init();
  tft.setRotation(2);
  tft.fillScreen(TFT_BLACK);

  progress_spr.setColorDepth(8);
  progress_spr.createSprite(240, 20);

  drawProgress(0, "Booting");
  delay(500);

  //init Wifi using WifiManager
  drawProgress(25, "Connecting to WiFi");
  delay(500);
  
  //use ESP8266 core NTP support
  drawProgress(50, "Getting time");
  delay(500);

  //attempt to get first reading
  drawProgress(75, "Getting data from ThingSpeak");
  delay(500);

  progress_spr.deleteSprite();

  tft.fillScreen(TFT_BLACK);
}

void loop(void)
{
  unsigned long currentMillis = millis();
  // fake update the display
  if ((currentMillis - lastUpdate) > 1000)
  {
    temp1++;
    temp2++;
    updateDisplay = true;
  }

  // Check if the display should be updated
  if (updateDisplay == true || firstRun)
  {
    firstRun = false;

    drawTemperature("Inside", temp1, 17.0, 25.0, 0, 30);
    drawTemperature("Outside", temp2, 17.0, 25.0, 0, 160);

    updateDisplay = false;
    lastUpdate = currentMillis;
  }

  delay(50);
}

// Progress bar helper
void drawProgress(uint8_t percentage, String text)
{
  Serial.println(text);

  progress_spr.fillSprite(TFT_BLACK);
  progress_spr.setFreeFont(&FreeSans9pt7b);
  progress_spr.setTextColor(TFT_WHITE, TFT_BLACK);
  progress_spr.setTextDatum(TC_DATUM);

  progress_spr.drawString(text, 120, 0);
  fex.drawProgressBar(10, 150, 240 - 20, 15, percentage, TFT_WHITE, TFT_BLUE);

  progress_spr.pushSprite(0, 120);
}

// Draw temperature blocks
void drawTemperature(String name, float temp, float temp_min, float temp_max, int poX, int poY)
{
  temp_spr.setColorDepth(8);
  temp_spr.createSprite(240, 130);
  temp_spr.fillSprite(TFT_BLACK);

  temp_spr.setTextDatum(TL_DATUM);
  temp_spr.setFreeFont(&FreeSans12pt7b);
  temp_spr.setTextColor(TFT_GREEN);
  temp_spr.drawString(name, 0, 0);

  temp_spr.setCursor(0, 60);
  temp_spr.setFreeFont(&FreeSansBold24pt7b);
  temp_spr.setTextColor(TFT_YELLOW);
  temp_spr.print(temp);
  temp_spr.print("`C");

  temp_spr.setFreeFont(&FreeSans12pt7b);
  temp_spr.setTextColor(TFT_LIGHTGREY);
  temp_spr.drawString("Min", 0, 72);
  temp_spr.drawString("Max", 120, 72);

  temp_spr.setFreeFont(&FreeSansBold18pt7b);
  temp_spr.setCursor(0, 120);
  temp_spr.setTextColor(TFT_CYAN);
  temp_spr.print(temp_min);
  temp_spr.print("`C");

  temp_spr.setCursor(120, 120);
  temp_spr.setTextColor(TFT_RED);
  temp_spr.print(temp_max);
  temp_spr.print("`C");

  temp_spr.pushSprite(poX, poY);
  temp_spr.deleteSprite();
}
```

`platformio.ini`
```
; 1467  -   JPEGDecoder (needed by TFT_eFEX)
; 1559  -   TFT_eSPI
; 6164  -   TFT_eFEX

[env:d1mini-serial]
platform = espressif8266
framework = arduino
board = d1_mini
lib_deps =
  1467
  1559
  6164
build_flags =
  -DUSER_SETUP_LOADED
  -DILI9341_DRIVER
  -DTFT_CS=15
  -DTFT_DC=0
  -DTFT_RST=-1
  -DLOAD_GLCD
  -DLOAD_FONT2
  -DLOAD_FONT4
  -DLOAD_FONT6
  -DLOAD_FONT7
  -DLOAD_FONT8
  -DLOAD_GFXFF
  -DSMOOTH_FONT
  -DSPI_FREQUENCY=27000000
  -DSPI_READ_FREQUENCY=20000000
  -DSPI_TOUCH_FREQUENCY=2500000
monitor_speed = 115200
```

Bodmer, 2019-07-26:
Glad to hear it is working!

closed 2019-07-26 by Bodmer