Bodmer/TFT_eSPI #400: setFreeFont(NULL) causes ESP32 crash

dracir9, 2019-07-21:
I have written something like this:
```
const GFXfont *fontvar = NULL;
*** Some code here that may change fontvar ***
tft.setFreeFont(fontvar);
```
If fontvar is set to one of the free fonts everything works properly but if it remains null then code crashes. I'd like to set text font to GLCD when fontvar is null.

In the Free_Fonts.h file in Free_Font_Demo example it says:
```
//  Setting the font to NULL will select the GLCD font:
//
//      tft.setFreeFont(NULL); // Set font to GLCD
```
But when using tft.setFreeFont(NULL) ESP32 crashes and keeps rebooting. Looking into the source code of that function I've seen that there is no check for whether GFXFont pointer is valid or not. It then tries to resolve pointer and crashes.

I could easily solve this by
```
if (font)
{
    tft.setFreeFont(fontvar);
}
else
{
    tft.setTextFont(GLCD);
}
```
It is not a big issue but I think I could help to maintain coherence.

Bodmer, 2019-07-21:
Thanks for pointing out this issue and the solution. I will correct the library in the next release.

Bodmer, 2019-07-30:
Thanks for reporting this, it should now be fixed. Please reopen issue if you find this is not the case.

closed 2019-07-30 by Bodmer