Bodmer/TFT_eSPI #401: Create a single Button

JeffTheMess, 2019-07-21:
I feel embarrassed for asking, I am very new to this world.
How do I create a single button? 
Once I have that I should be able to create everything I need from there. 
Sorry and thank you. (I couldn't find anything online for something I thought would be somewhere out there).

Bodmer, 2019-07-21:
I imported the Adafruit button class from the Adafruit_GFX library and added the capability to have button labels in any font. Examples and help are in short supply.

The tutorial [here](url) is relevant and I adapted the code as in [this example](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/320%20x%20240/Keypad_240x320).

The [example here](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Generic/On_Off_Button) does not use the button class and code to handle the buttons is in the sketch.

If you need more help then I will create a single button example but it will be a while before I will have time as I have a backlog of tasks!

JeffTheMess, 2019-07-21:
So I was able to create a single button from the example above. I'll add the code here. It made need some cleaning up and better syntax. Please let me know if this is good. It will be closed afterwards.
`#include "FS.h"

#include <SPI.h>
#include <TFT_eSPI.h>      // Hardware-specific library

TFT_eSPI tft = TFT_eSPI(); // Invoke custom library

// This is the file name used to store the calibration data
// You can change this to create new calibration files.
// The SPIFFS file name must start with "/".
#define CALIBRATION_FILE "/TouchCalData1"

// Set REPEAT_CAL to true instead of false to run calibration
// again, otherwise it will only be done once.
// Repeat calibration if you change the screen rotation.
#define REPEAT_CAL false

// Keypad start position, key sizes and spacing
#define KEY_X 120 // Center of key
#define KEY_Y 170
#define KEY_W 70 // Width and height
#define KEY_H 30
#define KEY_TEXTSIZE 1   // Font size multiplier

// Using two fonts since numbers are nice when bold
#define LABEL1_FONT &FreeSansOblique12pt7b // Key label font 1

// Create Signal Button
char keyLabel[] = "TEST";
uint16_t keyColor = TFT_BLUE;

// Invoke the TFT_eSPI button class and create all the button objects
TFT_eSPI_Button key;

//------------------------------------------------------------------------------------------

void setup() {
  // Use serial port
  Serial.begin(9600);

  // Initialise the TFT screen
  tft.init();

  // Set the rotation before we calibrate
  tft.setRotation(0);
  
  // Draw keypad background
  tft.fillRect(0, 0, 240, 320, TFT_DARKGREY);
  // Draw keypad
  drawKeypad();
}
//------------------------------------------------------------------------------------------

void loop(void) {
  uint16_t t_x = 0, t_y = 0; // To store the touch coordinates

  // Pressed will be set true is there is a valid touch on the screen
  boolean pressed = tft.getTouch(&t_x, &t_y);

  // / Check if any key coordinate boxes contain the touch coordinates
  if (pressed && key.contains(t_x, t_y)) {
    key.press(true);  // tell the button it is pressed
    Serial.println("TRUE");
    } else {
      key.press(false);  // tell the button it is NOT pressed
      Serial.println("FALSE");
    }
  
}
//------------------------------------------------------------------------------------------

void drawKeypad()
{
  // Draw the Button
      tft.setFreeFont(LABEL1_FONT);
      key.initButton(&tft,                                    // REF - LEAVE AS IS
                      KEY_X,                                  // X Cord: SEE ABOVE Line 19
                      KEY_Y,                                  // Y CORD: SEE ABOVE Line 20
                      KEY_W,                                  // WIDTH: SEE ABOVE Line 21
                      KEY_H,                                  // HEIGHT: SEE ABOVE Line 22
                      TFT_WHITE,                              // OUTLINE
                      keyColor,                               // TEXT COLOR
                      TFT_WHITE,                              // FILL
                      keyLabel,                               // TEXT TO PRINT
                      KEY_TEXTSIZE);                          // TEXT SIZE: SEE ABOVE Line 23
      key.drawButton();
    }`

closed 2019-07-26 by JeffTheMess

Bodmer, 2019-07-26:
Looks OK to me. The acid test is weather it works for you. If you need more help the do ask.

fungreenfox, 2021-12-18:
> 
> The tutorial [here](url) is relevant and I adapted the code as in [this example](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/320%20x%20240/Keypad_240x320).
> 
The link returns no result.