Bodmer/TFT_eSPI #402: percent sign leaving fragments

spyder0069, 2019-07-22:
Using a 128x128 1.44" display st7735 when I write a line such as 100% with the text background the same as the regular background and then write that same line as 0% with several additional spaces (used to clear the display in that area)" I get a corrupted % sign and some fragments still on the background.  This was with tft.setTextFont(2) if it matters.  For now it looks like i have to blank the space with a filled rectangle first in order to overcome this.

Bodmer, 2019-07-23:
The font is proportional and thus characters have different widths so padding with spaces does not work well.

Check out the setTextDatum() and setTextPadding() and use drawString().

spyder0069, 2019-07-23:
Thank you for the reply.  I tried the setTextPadding() and it unfortunately had the same issue for me.  It did clear the area but the percent sign is left corrupted.  You can see a couple of pixels lit on the right.  I don't have this issue with other characters.  Its as if the amount of space the percent sign character uses in the code is a pixel or two not wide enough so when the background overwrites it doesn't hit those pixels.  I will attach a pic.  I appreciate any advice

[image]

Bodmer, 2019-07-23:
Please provide a simple example sketch that demonstrates the problem.

spyder0069, 2019-07-23:
It looks like the problem is only when its set to the #2 font.  
tft.setTextFont(2);

Both of these work fine:
tft.setTextFont(1);
tft.setTextFont(4);

However if you adjust to the #2 font and adjust the padding statement to match then the percent gets the corruption.  This code recreates the problem:

`#include <TFT_eSPI.h> // Graphics and font library for ST7735 driver chip
#include <SPI.h>
TFT_eSPI tft = TFT_eSPI();  // Invoke library, pins defined in User_Setup.h

#define TFT_BLACK 0x0000
#define TFT_YELLOW 0xFFE0

#define TFT_CLK 18
#define TFT_MOSI 23
#define TFT_RST        4       
#define TFT_CS         5          
#define TFT_DC         33                                    
#define TFT_BACKLIGHT   32 

byte tft_padding;

void setup(void) {
  tft.init();
  tft.setRotation(2);
  tft.setTextFont(2);
  tft.setTextWrap(false);
  tft.fillScreen(TFT_BLACK);
  tft.setTextColor(TFT_YELLOW,TFT_BLACK);
  tft_padding = tft.textWidth("100%", 2);
  tft.setTextPadding(tft_padding);
}

void loop() {
  tft.drawString("100%",0,0);
  delay(2000);
  tft.drawString("99%",0,0);
  delay(2000);
}`

spyder0069, 2019-07-23:
Also if I just use my original method of using spaces to blank out extra characters it works if the font is set to 4 but fails with the same corruption on the percent sign if its set to 2.

Bodmer, 2019-07-24:
OK, I understand now. Font 2 is unique in the way the characters are specified and this produces a 1 pixel error in the width calculation.  I will correct this.

In the meantime this should work and will mean the % sign does not move:

'''
void loop() {
tft.setTextDatum(TR_DATUM);
tft.drawString("100%", tft.textWidth("100%", 2),0);
delay(2000);
tft.drawString("99%", tft.textWidth("100%", 2),0);
delay(2000);
}
'''

Bodmer, 2019-07-24:
Fixed in version 1.4.13

Thanks for reporting this issue.

spyder0069, 2019-07-24:
You are amazing and I appreciate your time and effort!  This did fix the method with using the padding.  Funny though it did not fix the method of using a space after the percent sign to blank any previous characters.  Seems like its must be along the same lines and again only affecting font 2.  I can use the padding method but it seems like just adding a blank character is an easier process if you don't need right justification.  

Here is the example:

`#include <TFT_eSPI.h> // Graphics and font library for ST7735 driver chip
#include <SPI.h>
TFT_eSPI tft = TFT_eSPI();  // Invoke library, pins defined in User_Setup.h

#define TFT_BLACK 0x0000
#define TFT_YELLOW 0xFFE0

#define TFT_CLK 18
#define TFT_MOSI 23
#define TFT_RST        4       
#define TFT_CS         5          
#define TFT_DC         33                                    
#define TFT_BACKLIGHT   32 

byte tft_padding;

void setup(void) {
  tft.init();
  tft.setRotation(2);
  tft.setTextFont(2);
  tft.setTextWrap(false);
  tft.fillScreen(TFT_BLACK);
  tft.setTextColor(TFT_YELLOW,TFT_BLACK);
  //tft_padding = tft.textWidth("100%", 2);
  //tft.setTextPadding(tft_padding);
}

void loop() {
  tft.drawString("100%",0,0);
  delay(2000);
  tft.drawString("99%  ",0,0);
  delay(2000);
}`

Bodmer, 2019-07-24:
I think the problem with using a space is that the character is narrower than the percent sign. Font 2 is proportional so the character widths vary. I have however made all digits the same width.

spyder0069, 2019-07-24:
I see what you are saying however if I use another character instead of the % it works.  Here is an example with different widths that still work:

`  tft.drawString("100d0",0,0);
  delay(2000);
  tft.drawString("99dd  ",0,0);
  delay(2000);`

It seems like the percent sign character is reserving an extra vertical pixel line as it builds the printed line but doesn't fill it.  So any character after it is being put one pixel further to the right and that skipped column isn't getting overwritten.  You can recreate the failure with this as well:
`  tft.drawString("100%0",0,0);
  delay(2000);
  tft.drawString("99%0  ",0,0);
  delay(2000);`

So its probably in the reported width of the font character?

spyder0069, 2019-07-24:
`tft.textWidth("%", 2);`

This does report the character length as 8 which is correct but the blank space that is being used after the percent sign to separate characters is not being drawn but is probably still being reserved in the line spacing.  Here is an example:

`  tft_padding = tft.textWidth("%", 2);  
  Serial.println("%=" + String(tft_padding));
  tft.drawString("%P ",0,0);
  delay(5000);
  tft_padding = tft.textWidth("W", 2);  
  Serial.println("W=" + String(tft_padding));
  tft.drawString("WP ",0,0);  
  delay(5000);`

In this case the W takes 9 spaces and the % only 8.  When it tries to overwrite it does create the space between the % and the P but it does not fill it and you are left with pixels left from the W.

[image]

spyder0069, 2019-07-24:
In this example I just filled with W and the overwrote with % and you can see its missing the filled blank space after each percent.
[image]

spyder0069, 2019-07-26:
Did those images help?

Bodmer, 2019-07-26:
I suspect this is a bug. The % sign is encoded as an 8 bit wide image to save space, but the spacing to the next character is 9 pixels. This means the character rendering has to add an extra pixel column. The issue therefore probably only affects 8 bit wide images characters and there is only 5 of those in the character set.

I will log this as a bug, reopen this issue and investigate further

Bodmer, 2019-07-30:
Thanks for reporting this, it should now be fixed. Please reopen issue if you find this is not the case.

closed 2019-07-30 by Bodmer

spyder0069, 2019-07-30:
Confirmed as fixed.  THANKS!!!  How do I donate for your time?

commit af4a96a4 (2019-07-24): Fix #402