Bodmer/TFT_eSPI #403: Compinling Errors since Update to v1.4.12

Holgiman, 2019-07-23:
Hello,

since update to version 1.4.12, yesterday I got several compiling errors.
Any hints how to solve or to download an older version of this lib?
Thanks in advance for any support.

|-- <TFT_eSPI> 1.4.12
|   |-- <SPI> 1.0
|-- <SPI> 1.0
Compiling .pio\build\esp12e\src\main.cpp.o
src\main.cpp: In function 'void setup()':
src\main.cpp:92:19: error: 'touch_calibrate' was not declared in this scope
   touch_calibrate();
                   ^
src\main.cpp:105:14: error: 'drawKeypad' was not declared in this scope
   drawKeypad();
              ^
src\main.cpp: In function 'void loop()':
src\main.cpp:114:25: error: 'class TFT_eSPI' has no member named 'getTouch'
   boolean pressed = tft.getTouch(&t_x, &t_y);
                         ^
src\main.cpp:143:18: error: 'status' was not declared in this scope
         status(""); // Clear the old status
                  ^
src\main.cpp:153:18: error: 'status' was not declared in this scope
         status(""); // Clear the old status
                  ^
src\main.cpp:157:43: error: 'status' was not declared in this scope
         status("Sent value to serial port");
                                           ^
src\main.cpp:163:31: error: 'status' was not declared in this scope
         status("Value cleared");
                               ^
src\main.cpp: In function 'void touch_calibrate()':
src\main.cpp:240:9: error: 'class TFT_eSPI' has no member named 'setTouch'
     tft.setTouch(calData);
         ^
src\main.cpp:259:9: error: 'class TFT_eSPI' has no member named 'calibrateTouch'
     tft.calibrateTouch(calData, TFT_MAGENTA, TFT_BLACK, 15);

Bodmer, 2019-07-23:
Probably your setup was over written and the touch controller chip select pin is no longer defined. Please check.

It is best to backup your setup files.

Holgiman, 2019-07-24:
I'm such a dumphead of a noob...  Yes, that's the reason.. Solved it now.
Thanks for fast response and help.
Greets from Germany

closed 2019-07-24 by Holgiman