Bodmer/TFT_eSPI #405: question: Speed Improvements, what optimations esp8266

tablatronix, 2019-07-24:
Hello , love this library!

I know the readme says a quick blurb about esp8266 optimizations over adafruit gfx, but I have looked through commits and code and am unsure what exactly they are.

Did you document those anywhere, I am curious where the 7x improvement I am observiing is coming from!

I swear I saw it at one time, maybe a blog post or issue?

Bodmer, 2019-07-26:
The speed improvements were done on and ad hoc basis during library development as new ideas for improvement of other features were implemented. I have not documented them. The library uses direct register access and uses the SPI buffers. It is also possible to optimise single pixel draws in the case where successive ones are in an orthogonal positions because the TFT accepts fewer bounding box commands.  The block fill function is used a lot by library functions and so has been optimised. See comments [here](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L4958). Some displays also work with high SPI clock rates which is a benefit. The line drawing algorithm has also been optimised a little.

closed 2019-07-26 by Bodmer

tablatronix, 2019-07-27:
Thanks, I figured as much, not something that can be PRed or patched to adafruit lib, Do you know what the single best improvement was? I am guessing draw rect optimizations and dma full buffer transfers

Bodmer, 2019-07-29:
The most significant improvement is using the SPI buffer as the content can be resent multiple times after filling, great for a block fill.

I have not tested the Adafruit library for quite a while.  I have been told they have made some improvements, so I suspect the performance gap has narrowed in recent times.

I doubt using DMA will help that much when you have a high clock rate processor, and may well end up being a slower approach for short pixel bursts which are often needed during display updates.