Bodmer/TFT_eSPI #407: German Umlauts not displayed correctly

Petros144, 2019-08-06:
Hi, im working on a project that gets Serial data and displays it with the ESP32.

I already made my own GFX Font with the extended ascii Table that includes (ÖÄÜ öäü ß)

I Checked the Font file with the Online viewer  https://tchapi.github.io/Adafruit-GFX-Font-Customiser/. (all ok)

I checked the Serial data, everything is right. it must be the Font file or Lib.

I check with this:

  tft.fillScreen(TFT_MAGENTA);            // Clear screen
  tft.setFreeFont(CF_AB30);                   // Select the font
  tft.drawString("Ö Ä Ü ö ä ü ß hello", 160, 60, GFXFF);// Print the string name of the font

And the only thing printed on the Screen is "ö hello"

if I Print :

tft.drawString("hä hö hü", 160, 60, GFXFF);// Print the string name of the font

I get:

"hö hü" on the display...

Where is the fault, is it the Font file or the Lib? 

current version Im useing is 1.4.8

Font file is attached

[c__windows_fonts_calibrib14pt8b.zip](https://github.com/Bodmer/TFT_eSPI/files/3473743/c__windows_fonts_calibrib14pt8b.zip)

closed 2019-08-07 by Petros144

Bodmer, 2019-08-07:
Seems to work for me. I tried attached sketch with version 1.4.16 of TFT_eSPI.

[Petros144.zip](https://github.com/Bodmer/TFT_eSPI/files/3474708/Petros144.zip)

Dakkaron, 2025-01-13:
Is this possible for the default font too?

From what I can tell, the upstream version of it should support Umlaut characters: https://github.com/adafruit/Adafruit-GFX-Library/blob/master/glcdfont.c