Bodmer/TFT_eSPI #408: UTF8 to Extended ASCII Decode

Petros144, 2019-08-07:
Hi Bodmer & comunity! 

Your Lib works great!
Im having some problems with UTF8 and displaying.
I basicly get Phone messanges (whatsapp, iMessage etc.) and often there are Emojis in there,
is it possible to detect them and only display the extended ASCII code (for umlauts)?

I tried the "Arduino UTF8 Decoder" but it wont work on the ESP32 somehow?

https://playground.arduino.cc/Main/Utf8ascii/

This worked fine on Atmega328 but let the ESP restart...

I saw that your Library have some utf8 decoding in it, but i cant understand how it works or if it fits my needs.

Bodmer, 2019-08-08:
There is two functions in the library to decode UTF8. One handles arrays and the other a serial stream and the calling program would use one of these. The choice is dependant on whether you already have the message stings in a character array or whether the bytes are arriving serially.

The library decoders only work with 16 bit Unicode point codes, so if the Emoji use 24 bit point codes then the library will not work for you. However it would be possible to copy a function to your sketch and adapt it.

So to help I need more information.

Petros144, 2019-08-08:
Hi Bodmer! apreciate the help!

So my data comes from a Feather nRF52832  and get it transmited via Serial:
I´ve got the UTF8 Arduino decoder to work on the Feather, now i recivce the extended ASCII in the Serial monitor without Unicode.

I attach the Console output,  looks like the Hex values are right:
[image]

The problem is now , that the ESP won´t recognize every character properly and wont print them at all.

this is how I process the Serial Stream on the ESP to save my variables

`  while (Serial2.available()  ) // Variablen einlesen
  {

    Name  = Serial2.readStringUntil('\n');  // This is the Name I Display

    Nachricht = Serial2.readStringUntil('\n');  // **<<<<<<This is the Message I Display**

    Kategorie  = Serial2.readStringUntil('\n');

    StundeBle  = Serial2.readStringUntil('\n');

    MinuteBle  = Serial2.readStringUntil('\n');

    SekundeBle  = Serial2.readStringUntil('\n');

    JahrBle  = Serial2.readStringUntil('\n');

    MonatBle  = Serial2.readStringUntil('\n');

    TagBle  = Serial2.readStringUntil('\n');
}`

The code for output looks like this: 

          //### Text Mesasge ###
          tft.drawCentreString(lineOne, 160, 45, GFXFF);
          tft.drawCentreString(lineTwo, 160, 72, GFXFF);
          tft.drawCentreString(lineThree, 160, 99, GFXFF);
          tft.drawCentreString(lineFour, 160, 126, GFXFF);
          tft.drawCentreString(lineFive, 160, 153, GFXFF);
          tft.drawCentreString(lineSix, 160, 180, GFXFF);
          tft.drawCentreString(lineSeven, 160, 207, GFXFF);

The String must be " **öäü ÖÄÜß hello**"

this is the result:

[image]

is it possible that the String class is the problem? 

some more examples:

Message: hellö hellä
console : hellö hellä
TFT: hellö hellä (all good)
----------------

Message: hellö hellä hellü
TFT: hellö hellellü ( h & ä are removed somehow)

if i print something hardcoded it shows up correctly, it must be something with the encoding. this freaks me out.

Bodmer, 2019-08-08:
The problem appears to be that your input stream is not actually UTF8 encoded. It is using extended ASCII (character codes in range 0x80 to 0xFF).

You can read about UTF8 coding [here](https://en.wikipedia.org/wiki/UTF-8), in particular see the Codepage layout section. Not that most of the extended ASCII codes are used as part of the UTF8 coding so these are normally decoded to create a 16 bit point code.

The Arduino utf8ascii code does not appear to be a proper UTF8 decoder, it just extracts the 8 bit ascii values plus the euro symbol.

You can use [this online tool](https://mothereff.in/utf-8) to see what the UTF8 encoded byte sequence should look like. For example: öäü ÖÄÜß hello will look like:

\xC3\xB6\xC3\xA4\xC3\xBC\x20\xC3\x96\xC3\x84\xC3\x9C\xC3\x9F\x20\x68\x65\x6C\x6C\x6F\x0A

Petros144, 2019-08-08:
The main goal is to display characters without the emoji stuff.

The UTF8 decoder does the job but gives me this unprintable Extended ASCII that the Library not print/display corectly- this won´t work.

is it better to handle the RAW data from the Feather (no decoding and proper UTF8) and proccess it in the ESP32? - maybe with your decoder?

This is the Raw data with UTF8:

[image]

and it prints totaly fine on the ILI9341

[image]

But now I stuck at the same problem as before, the Unicode characters are messing it up:

[image]

this is with a emoji in it:

[image]

and

[image]

How can I filter them out? is there a way to do this?

I know that this discussion goes a bit off topic. but i really stuck at this last problem to complete my project.

Bodmer, 2019-08-09:
I don't mind the off topic discussion!

The emoji are 21 bit code points which are not supported by TFT_eSPI and the decoder state machine is not handling them. For codes that do not exist in the character set loaded, by convention, boxes are drawn to signal a missing character.

If I understand correctly you have 2 processors, the first in the chain being a feather board. In that case there are 2 options:
1. Filter out 21 bit codes in the feather but do bot convert to ASCII
2. Update of TFT_eSPI

Option 1 is the best option in the short ter to get your project back on track. What you need to do is look at each character (call it variable c) and see if ( (c & 0xF8) == 0xF0) ) then throw away the next 3 characters if, for each of these three, this is true ( (c & 0xC0) == 0x80) ). What you are trying to do is filter out the last line in the table:

[image]

Eventually I will update TFT_eSPI to handle 21 bit code points. Normally I do more coding/testing in the winter so it will be a while before I reserve time for that. I will also need to add in some attribute switches to control whether boxes are drawn. I will also add a switch to turn on/off UTF8 decoding so the library will handle extended ASCII, there is already a setAttribute() function for this but currently this is undocumented and only works with the print stream not drawString().

Petros144, 2019-08-09:
Now you lead me to the right direction.

You are correct, the Feather MCU collect´s the data and send´s it to the ESP with the TFT_eSPI Lib.

I am not very experienced with handlig of the Char arrays and I am sorry that I am so uncertain instead.
I probably need a starting point to continue to work this out myself somehow.
This is my feather Code right now, it runs Apple Notification Service Via BLE:
This is made from the ANCS example.

`
//Var declarations

char buffer[128];

#define BUFSIZE     128

char title[BUFSIZE];
char message[BUFSIZE];
char app_name[BUFSIZE];
char category[BUFSIZE];
char datetime[BUFSIZE];

int jahr;
int monat;
int tag;
int stunde;
int minuten;
int sekunden;

String title_format;
String message_format;

void ancs_notification_callback(AncsNotification_t* notif)
{

  if (notif->eventID == ANCS_EVT_NOTIFICATION_ADDED )
  {

    bleancs.getAttribute(notif->uid, ANCS_ATTR_TITLE, title , BUFSIZE);           //<<< This is what is displayed
    bleancs.getAttribute(notif->uid, ANCS_ATTR_MESSAGE, message , BUFSIZE);       //<<< This is what is displayed
    bleancs.getAttribute(notif->uid, ANCS_ATTR_APP_IDENTIFIER, app_name, BUFSIZE);
    bleancs.getAttribute(notif->uid, ANCS_ATTR_DATE , datetime, BUFSIZE);

    sscanf(datetime, "%4d%2d%2dT%2d%2d%2d", &jahr, &monat, &tag, &stunde, &minuten, &sekunden);

    //    utf8ascii(title);
    title_format = title;
    title_format.replace('\n', ' ');

    //     utf8ascii(message);
    message_format = message;
    message_format.replace('\n', ' ');

    // PRINT TO SERIAL 1//

    Serial.print(title_format);
    Serial.print('\n');
    Serial.print(message_format);
    Serial.print('\n');
    Serial.print(notif->categoryID);
    Serial.print('\n');
    Serial.print(stunde);
    Serial.print('\n');
    Serial.print(minuten);
    Serial.print('\n');
    Serial.print(sekunden);
    Serial.print('\n');
    Serial.print(jahr);
    Serial.print('\n');
    Serial.print(monat);
    Serial.print('\n');
    Serial.print(tag);
    Serial.print('\n');

    // PRINT TO SERIAL 1// END

    uid = notif->uid;

  }

}`

So I use char message[BUFSIZE] 

1. do i need to know how many chars are in there or do i use the full buffer of 128?

2. I make a for loop and compare each char.

(this is totaly wrong, I don´t know how to compare single chars from this Buffsize array)

  for (int i = 0; i < BUFSIZE; i++)
  {
    c = message[BUFFSIZE];
   if (c & 0xF8) == 0xF0) 
{
i+=2;
}
  }

my knowledge is not enough to solve this problem myself :shame:

EDIT:

Also I found this in the Sketch:

```
    // iDevice often include Unicode "Bidirection Text Control" in the Title.
    // Mostly are U+202D as beginning and U+202C as ending. Let's remove them
    if ( bleancs.getTitle  (uid, myNtf->title   , BUFSIZE) )
    {
      char u202D[3] = { 0xE2, 0x80, 0xAD }; // U+202D in UTF-8
      char u202C[3] = { 0xE2, 0x80, 0xAC }; // U+202C in UTF-8

      int len = strlen(myNtf->title);

      if ( 0 == memcmp(&myNtf->title[len-3], u202C, 3) )
      {
        len -= 3;
        myNtf->title[len] = 0; // chop ending U+202C
      }

      if ( 0 == memcmp(myNtf->title, u202D, 3) )
      {
        memmove(myNtf->title, myNtf->title+3, len-2); // move null-terminator as well
      }
    }
```

may this help?

Bodmer, 2019-08-09:
You appear to already have the Arduino Utf8ascii code integrated into your Feather sketch, so the simplest way forward is to adapt the utf8ascii(byte ascii) function to do the filtering. Maybe this would work (untested!):

```
static byte c1;  // Character count to remove

// Modified to strip out 21 bit Unicode code points
byte utf8ascii(byte ascii) {
    if ( ascii<128 )   // Standard ASCII-set 0..0x7F handling  
    {   c1=0;
        return( ascii );
    }

   if ( (ascii & 0xF8) == 0xF0)  { c1 = 3; return 0; }

   else if (c1 && ( (c & 0xC0) == 0x80) )  ) { c1--; return 0; }

    c1 = 0;
    return  ascii);
}
```

Of course the function name does not reflect what it does anymore but this function should be a plug in replacement for the existing Arduino code you have running on the Feather.

Petros144, 2019-08-10:
Mister Bodmer,

You are a Genius! I really need to test the whole thing but the Display of the characters work perfect!
Thank you so much for your time & effort for the Lib and the Support here!

If you have a Donation Mail for PayPal or so, I will definitely put some Bucks in it! 

Thanks alot, really!

PS: Maybe add this to the Lib aswell?

Bodmer, 2019-08-12:
Glad to hear it is working. I don't take payments for help, if you so wish then make a donation to a charity of your choice.

closed 2019-08-12 by Bodmer