Bodmer/TFT_eSPI #409: No touch with ili9488 and xpt2046

TA4TA40S, 2019-08-07:
Here is my display: https://fr.aliexpress.com/item/32985467436.html?spm=a2g0o.productlist.0.0.fc0e39071Q5kwl&algo_pvid=62e81748-f7bd-4dcf-8395-ba0ba6ad326f&algo_expid=62e81748-f7bd-4dcf-8395-ba0ba6ad326f-1&btsid=4bc2d3c1-2b28-40ce-8a67-5540eab969e7&ws_ab_test=searchweb0_0,searchweb201602_1,searchweb201603_53
Here is my configuration:
#define ILI9488_DRIVER
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15
#define TFT_DC    4
#define TFT_RST   2  
#define TOUCH_CS 21
I didn't connect the screen SDO and merged all SPI lines (except CS) and didn't connect T_IRQ. When I run Touch_Controller_Demo I just have a white screen. When I run the 320*480 keypad I have the image but the touch doesn't work. In the other examples the lcd works fine.

Bodmer, 2019-08-07:
Run the [test sketch here](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Test%20and%20diagnostics/Read_User_Setup) and post the serial output here. This will output the settings picked up by the compiler.

Run the [touch test sketch](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Test%20and%20diagnostics/Test_Touch_Controller) and look at the serial output to see if the touch controller is responding.

TA4TA40S, 2019-08-08:
Output of the first sketch:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9296
load:0x40080400,len:6364
entry 0x400806a4

[code]
TFT_eSPI ver = 1.4.5
Processor    = ESP32
Frequency    = 240 MHz
Transactions = Yes 
Interface    = SPI 
Display driver = 9488
Display width  = 320 
Display height = 480 

MOSI    = D23 (GPIO 23)
MISO    = D19 (GPIO 19)
SCK     = D18 (GPIO 18)
TFT_CS   = D15 (GPIO 15)
TFT_DC   = D4 (GPIO 4)
TFT_RST  = D2 (GPIO 2)

TOUCH_CS = D21 (GPIO 21)

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.0 MHz 
Touch SPI frequency   = 2.5 MHz 
[/code]
With the second sketch the touch seems to work but when I don't touch the display the coordinates are around x: 0     y: 3702     z: 28  and I can't get a coordinate of 0 (in x, y or z).

TA4TA40S, 2019-08-08:
Edit: I just didn't use the calibrate sketch (sorry for the inconvenience)

closed 2019-08-08 by TA4TA40S