Bodmer/TFT_eSPI #412: Smooth Font Without Background

BleepBlonk, 2019-08-08:
I've noticed that if i have a background color the smooth font shows nicely as it should, but if i print it without a background color it doesn't look right. For example if i want to print over whats already on the screen it looks blocky and jagged like the default fonts enlarged.

Bodmer, 2019-08-09:
Look at examples Font_demo_1-4 and read the comments to see different methods of rendering smooth fonts. The edges of the characters must be blended with the bacground so you always need to specify the background colour.

closed 2019-08-09 by BleepBlonk

Bodmer, 2019-08-09:
The library does not currently allow aliasing over "on screen images" since this would require reading the pixel colour for each antialiased character pixel to derive the blended colour. So if that is what you were asking about then this is not currently supported.

BleepBlonk, 2019-08-10:
That is what I was looking for, would have made great eye candy but I can always work around it. Thanks for the lib and reply!