Bodmer/TFT_eSPI #413: Difficulty with Wemos D1 Mini TFT Touch Shield  - Works with Adafruit_Gfx

jpasqua, 2019-08-10:
I'm confident that this is my error, but I'm having difficulty finding it. I have a Wemos D1 Mini mounted on a 2.4" Touch Shield:
https://wiki.wemos.cc/products:d1_mini_shields:tft_2.4_shield

The display and touch controller work well (though slowly) with Adafruit_GFX and the ILI9341 driver. I have been unable to get anything but a white screen when using it with TFT_eSPI. I've tried several of the supplied examples. I have used the TFT_eSPI library successfully with a D1 Mini connected to a 2.8" TFT, but not via the shield.

Here is the setup file I am using and the settings correspond to those I'm using with my successful Adafruit_GFX test:

    // See SetupX_Template.h for all options available

    #define ILI9341_DRIVER

    #define TFT_CS D0  //for D1 mini or TFT I2C Connector Shield (V1.1.0 or later)
    #define TFT_DC D8  //for D1 mini or TFT I2C Connector Shield (V1.1.0 or later)
    #define TFT_RST -1 //for D1 mini or TFT I2C Connector Shield (V1.1.0 or later)
    #define TS_CS D3   //for D1 mini or TFT I2C Connector Shield (V1.1.0 or later)

    #define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
    #define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
    #define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
    #define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
    #define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
    #define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
    #define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

    #define SMOOTH_FONT

    //#define SPI_FREQUENCY       27000000
    #define SPI_FREQUENCY       40000000
    #define SPI_READ_FREQUENCY  20000000
    #define SPI_TOUCH_FREQUENCY  2500000

    //#define SUPPORT_TRANSACTIONS

I've tried this with and without SUPPORT_TRANSACTIONS and with variations of the SPI_FREQUENCY. I am reading the settings back using Read_User_Setup and the values look good:

    TFT_eSPI ver = 1.4.16
    Processor    = ESP8266
    Frequency    = 80 MHz
    Voltage      = 3.33 V
    Transactions = No 
    Interface    = SPI 
    SPI overlap  = No 
    
    Display driver = 9341
    Display width  = 240 
    Display height = 320 
    
    TFT_CS   = D0 (GPIO 16)
    TFT_DC   = D8 (GPIO 15)
    Font GLCD   loaded
    Font 2      loaded
    Font 4      loaded
    Font 6      loaded
    Font 7      loaded
    Font 8      loaded
    Smooth font enabled
    
    Display SPI frequency = 40.0 MHz 

In summary:
* The D1 Mini, Shield, and Display all work together using Adafruit_GFX.
* I have successfully used TFT_eSPI with a D1 Mini and a separate display (not through the shield)
* I am reading the user settings back and they are what I set them to.
* I'm confident this is an error on my part and would appreciate any guidance

jpasqua, 2019-08-10:
Ugh. Sorry to have bothered everyone. My setup file used `D0, D8, D3` instead of `PIN_D0, PIN_D8, PIN_D3`. This is an important distinction since the former are variable definitions and the latter are #define's. This screws up the way they are interpreted by TFT_eSPI.h.

I hope my error saves someone else some time.

closed 2019-08-10 by jpasqua

Bodmer, 2019-08-12:
Unfortunately the Arduino Dx designations do not always work for some reason and can map the signals to the wrong pins.  Hence the need to use PIN_Dx, as noted in the setup file.