Bodmer/TFT_eSPI #414: Text flicker and chart flicker

ThatBigPrint, 2019-08-16:
i hope you can help me and my `friend` are doing a project that requires us to update the 240x240 lcd and what ever we try we are not able to get rid of the flicker can you help us fix this ?

most of the code has been ported from our oled version and we have tried rectangles etc ata loss would appreciate a fix or help 

being run on an eps 32

this is the arduino code 

 `#include <Arduino.h>
#include <SPI.h>
#include <TFT_eSPI.h> // Graphics and font library for ST7735 driver chip
#include <Wire.h>

TFT_eSPI tft = TFT_eSPI();  // Invoke library, pins defined in User_Setup.h

#define ADC_BITS    12
//U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SH1106_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ PB6, /* data=*/ PB5, /* cs=*/ PB10, /* dc=*/ PA2, /* reset=*/ PA3);
//U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ PB6, /* data=*/ PB5, /* cs=*/ PB10, /* dc=*/ PA2, /* reset=*/ PA3);
//U8G2__MIDAS_128X128_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ PB13, /* data=*/ PB15, /* cs=*/ PB12, /* dc=*/ PB14, /* reset=*/ PA8);

int count = 0;
const int cycleDelay = 20;      // Change to adjust cycle time

// offset var (multiply by 100)
int offset = 0;
String stat;
const int numReadings = 10;
int mode = 1;
// 1 = Bar
// 2 = PSI
// 3 = kPa

//int Bar = 0;
int Bar = 1;
int PSI = 0;
int kPa = 0;

int boostPressure;
int boostMax = 0;
int boostMin = 0;

int readings[numReadings];
int readIndex = 0;
int total = 0;
int average = 0;

unsigned long startMillis;
unsigned long currentMillis;
const unsigned long period = 5;

const int sensorHistoryLength = 240;
int sensorHistory[sensorHistoryLength];
int sensorHistoryPos = sensorHistoryLength - 1;

bool injector = false;          // are we using injectors
float ONCycle = 0;              // oncycle variable 
float OFFCycle = 0;             // offcycle variable got microsecond
float T = 0;                    // tota l time to one cycle ONCycle + OFFcycle
int F = 0;                      // Frequency = 1/T
int RPM = 0;                    // RPM (Frequency * 60)
float DutyCycle = 0;            // D = (TON/(TON+TOFF))*100 %
float dutyMax = 0;
float onMin = 0;

void setup(void) {
  tft.init();
  tft.setRotation(4);
  tft.fillScreen(TFT_BLACK);
[... 718 lines cut ...]
// #define SPI_FREQUENCY  80000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000

// The ESP32 has 2 free SPI ports i.e. VSPI and HSPI, the VSPI is the default.
// If the VSPI port is in use and pins are not accessible (e.g. TTGO T-Beam)
// then uncomment the following line:
#define USE_HSPI_PORT

// Comment out the following #define if "SPI Transactions" do not need to be
// supported. When commented out the code size will be smaller and sketches will
// run slightly faster, so leave it commented out unless you need it!

// Transaction support is needed to work with SD library but not needed with TFT_SdFat
// Transaction support is required if other SPI devices are connected.

// Transactions are automatically enabled by the library for an ESP32 (to use HAL mutex)
// so changing it here has no effect

// #define SUPPORT_TRANSACTIONS
`

Bodmer, 2019-08-18:
Your code is jumbled by formatting issues. Please create a zip of you sketch and drag to the comment window to upload.

ThatBigPrint, 2019-08-18:
[WORKING_1.4_atempt_lcd 2.zip](https://github.com/Bodmer/TFT_eSPI/files/3513466/WORKING_1.4_atempt_lcd.2.zip)

ThatBigPrint, 2019-08-18:
also there will be a chart displayed but we are not sure how we can update this and the non static text ie pressure readings etc my friend mentioned we will need to keep the previus reading in buffer and turn it white ?

ThatBigPrint, 2019-08-19:
any updates please ? we are stumped still

Bodmer, 2019-08-19:
I have run your example code and there are multiple issues:

1. Text leaving remnants on the  screen - use tft.setTextPadding() and setTextDatum() functions, see examples
2. High screen update rate - humans cannot read rapidly updated numbers, add a delay(250) at the end of the loop.

As I am busy with paid work at the moment I have very limited spare time to support your sketch development. Look at the examples and see how they work.

closed 2019-08-23 by Bodmer