Bodmer/TFT_eSPI #416: Getting many Warnings during compile with version 1.4.16

hasenradball, 2019-08-23:
Hi with the latest version I get many warning during compile...

```
Compiling library "TFT_eSPI"
"C:\\Users\\Frank\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Frank\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.0/tools/sdk/include" "-IC:\\Users\\Frank\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.0/tools/sdk/lwip2/include" "-IC:\\Users\\Frank\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\Frank\\AppData\\Local\\Temp\\arduino_build_456575/core" -c -Wall -Wextra -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -fno-exceptions -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10809 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\Frank\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.0\\cores\\esp8266" "-IC:\\Users\\Frank\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.0\\variants\\nodemcu" "-IC:\\Users\\Frank\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.0\\libraries\\ESP8266WebServer\\src" "-IC:\\Users\\Frank\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.0\\libraries\\ESP8266WiFi\\src" "-ID:\\06-1_Arduino\\libraries\\DHT_sensor_library_for_ESPx" "-ID:\\06-1_Arduino\\libraries\\TFT_eSPI" "-IC:\\Users\\Frank\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.0\\libraries\\SPI" "D:\\06-1_Arduino\\libraries\\TFT_eSPI\\TFT_eSPI.cpp" -o "C:\\Users\\Frank\\AppData\\Local\\Temp\\arduino_build_456575\\libraries\\TFT_eSPI\\TFT_eSPI.cpp.o"
D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:284:6: warning: unused parameter 'tc' [-Wunused-parameter]

 void TFT_eSPI::init(uint8_t tc)

      ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:791:6: warning: unused parameter 'mask' [-Wunused-parameter]

 void busDir(uint32_t mask, uint8_t mode)

      ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:791:6: warning: unused parameter 'mode' [-Wunused-parameter]

In file included from C:\Users\Frank\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/pgmspace.h:7:0,

                 from D:\06-1_Arduino\libraries\TFT_eSPI/Fonts/glcdfont.c:6,

                 from D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.h:62,

                 from D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:17:

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'int16_t TFT_eSPI::textWidth(const char*, uint8_t)':

C:\Users\Frank\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:74:85: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]

     #define pgm_read_dword(addr)            (*reinterpret_cast<const uint32_t*>(addr))

                                                                                     ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:2346:32: note: in expansion of macro 'pgm_read_dword'

     char *widthtable = (char *)pgm_read_dword( &(fontdata[font].widthtbl ) ) - 32; //subtract the 32 outside the loop

                                ^

C:\Users\Frank\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:74:85: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]

     #define pgm_read_dword(addr)            (*reinterpret_cast<const uint32_t*>(addr))

                                                                                     ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:2369:45: note: in expansion of macro 'pgm_read_dword'

           GFXglyph *glyph  = &(((GFXglyph *)pgm_read_dword(&gfxFont->glyph))[uniCode]);
```

Bodmer, 2019-08-23:
Try updating to latest 2.5.2 core.

hasenradball, 2019-08-24:
Ok thank you I will check it.
I am using 2.5.0

hasenradball, 2019-08-24:
Hi,
I now changed the core to 2.5.2 and I getting just these warnings:
```
D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:284:6: warning: unused parameter 'tc' [-Wunused-parameter]

 void TFT_eSPI::init(uint8_t tc)

      ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:791:6: warning: unused parameter 'mask' [-Wunused-parameter]

 void busDir(uint32_t mask, uint8_t mode)

      ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:791:6: warning: unused parameter 'mode' [-Wunused-parameter]

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::setWindow(int32_t, int32_t, int32_t, int32_t)':

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:2713:22: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]

   SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);

                      ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:2713:47: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]

   SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);

                                               ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:2734:22: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]

   SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);

                      ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:2734:47: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]

   SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);

                                               ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:2751:22: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]

   SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);

                      ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:2751:47: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]

   SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);

                                               ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::readAddrWindow(int32_t, int32_t, int32_t, int32_t)':

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:2971:22: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]

   SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);

                      ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:2971:47: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]

   SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);

                                               ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:2988:22: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]

   SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);

                      ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:2988:47: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]

   SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);

                                               ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:3005:22: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]

   SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);

                      ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:3005:47: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]

   SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);

                                               ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'virtual void TFT_eSPI::drawPixel(int32_t, int32_t, uint32_t)':

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:3082:24: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]

     SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);

                        ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:3082:49: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]

     SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);

                                                 ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:3109:24: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]

     SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);

                        ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:3109:49: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]

     SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);

                                                 ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:3134:22: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]

   SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);

                      ^

D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:3134:47: warning: suggest parentheses around '-' inside '<<' [-Wparentheses]

   SPI1U1 = (CMD_BITS << SPILMOSI) | (CMD_BITS << SPILMISO);

                                               ^

In file included from D:\06-1_Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:5364:0:

D:\06-1_Arduino\libraries\TFT_eSPI\Extensions/Sprite.cpp: In member function 'void TFT_eSprite::pushImage(int32_t, int32_t, int32_t, int32_t, const uint16_t*)':

D:\06-1_Arduino\libraries\TFT_eSPI\Extensions/Sprite.cpp:719:31: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]

       for (int32_t xp = 0; xp<ww; xp+=8)
```

Bodmer, 2019-08-24:
Those should be just warnings and should not stop the sketches compiling and running.

closed 2019-08-24 by Bodmer

hasenradball, 2019-08-24:
Yes but a good and clean code has also no warnings 😉

hasenradball, 2019-08-24:
If I can work it out to make it clean can I make a PR?

Bodmer, 2019-08-24:
I have already corrected them in my copy according to this PR:
https://github.com/Bodmer/TFT_eSPI/pull/389

but will not be uploadinf for a while.