Bodmer/TFT_eSPI #417: Transparent sprite boxes results in colored background boxes

Joennuh, 2019-08-23:
I'm quite new into Arduino but with some experience on PHP and a short course on C# I do have some knowledge about the basics.

I currently have a LilyGo TTGO T-display with ST7789 IPS color display on which I want to try to build a little weatherdisplay combined with attached sensors like DS18B20. So far driving the display is going well but there is 1 thing I can't get working in the right way:

I want to use a sprite to update the text on the display. Currently I create a 150 x 50 pixels sprite and fill it with `fillSprite` and `TFT_GREEN` as color.

For creating the text I use my own font, set the foreground color in `drawString` to `TFT_WHITE` and the backgroundcolor to `TFT_GREEN`. As I understand this is only needed to generate a correct anti-aliased font and the `TFT_GREEN` in `drawString` should be rendered as transparent.

Then I use `pushSprite` and provide `TFT_GREEN` as color. As I understand this should render the `TFT_GREEN` used at `fillSprite` transparent.

Now the issue is that I see a red background behind my text instead of the black background `TFT_BLACK` which I provided at the beginning of my sketch with `fillScreen(TFT_BLACK);`.

Is this a bug or is it something that I have missed?

Below is not the actual sketch but what I think the most important parts of the actual sketch:
```c++
#include <FS.h>
#include <SPI.h>
#include <TFT_eSPI.h>
#include <Button2.h>
#include "bmp.h"

#ifndef TFT_DISPOFF
#define TFT_DISPOFF 0x28
#endif

#ifndef TFT_SLPIN
#define TFT_SLPIN   0x10
#endif

#define TFT_MOSI            19
#define TFT_SCLK            18
#define TFT_CS              5
#define TFT_DC              16
#define TFT_RST             23

#define LED_TFT 4

TFT_eSPI gfx = TFT_eSPI(135, 240);
TFT_eSprite spr = TFT_eSprite(&gfx); // Sprite class needs to be invoked

void setup(){
	pinMode(LED_TFT, OUPUT);
	digitalWrite(LED_TFT, HIGH);
	
	gfx.init();

	gfx.setRotation(1);

	spr.setColorDepth(16); // 16 bit colour needed to show antialiased fonts

	gfx.setTextWrap(false);
	gfx.fillScreen(TFT_WHITE);

	delay(250);

	//Serial.print("Showing bootlogo... ");
	gfx.setSwapBytes(true);
	gfx.pushImage(0, 0,  240, 135, bootlogo);
	delay(500);
	gfx.fillScreen(TFT_BLACK);
}
  
void loop(){
	spr.loadFont("OpenSans36");
	spr.createSprite(150, 50);
	spr.fillSprite(TFT_GREEN);
	spr.drawRect(0, 0, 150, 50, TFT_WHITE);
	spr.setTextColor(TFT_WHITE, TFT_GREEN); // Set the sprite font colour and the background colour
	spr.setTextDatum(MC_DATUM); // Middle Centre datum
	spr.drawString(String("25.2") + String("\xB0") + String("C"), 75, 25 ); // Coords of middle of 150 x 50 Sprite
	spr.pushSprite(10, 10, TFT_GREEN); // Push to TFT screen coord 10, 10. Blue is transparent
	spr.unloadFont(); // Remove the font from sprite class to recover memory used
	spr.deleteSprite();
	delay(1000);
}
```

Bodmer, 2019-08-24:
Thanks for reporting this. I seem to have introduced a bug.

You can get around this in the short term by adding this line to setup():
  gfx.setSwapBytes(false);

And then using TFT_BLACK as the transparent colour.

I will try to find a solution in the next week or so.

Joennuh, 2019-08-24:
The workaround does help a little bit: in the main loop I see less issues regarding colors. But in the `setup()` function I now have a horrible looking splash screen. It should be a few red filled circles with a gradient but they are now ugly blue. 😢 
It is just a personal project and it does not have a deadline. So I will try building the GUI and wait for the bugfix.
Thank you for your effort taken so far!

Bodmer, 2019-08-24:
Further tests show you can use any transparent colour, you just nee to put:

gfx.setSwapBytes(false);

before the pushSprite()

and:
gfx.setSwapBytes(true);

This is due to endianess issues.

Joennuh, 2019-08-24:
Yes, that seem to help a lot. With that workaround it behaves almost as it should be.
The only frustrating thing is that I can't draw a symbol with black in it:
```c++
// Icons upper right corner
      sprIcons.createSprite(70, 22);
      spr.fillSprite(TFT_BLACK);
      sprIcons.drawXBitmap(0, 2, iconLedOutline, 16, 18, TFT_WHITE);
      if(torch>0){
        sprIcons.drawXBitmap(0, 2, iconLedFill, 16, 18, TFT_WHITE);
      }
      else{
        sprIcons.drawXBitmap(0, 2, iconLedFill, 16, 18, TFT_BLACK);
      }
      gfx.setSwapBytes(false);
      sprIcons.pushSprite(170,0,TFT_BLACK);
      gfx.setSwapBytes(true);
      sprIcons.deleteSprite();
```
When `torch` is more then `0` it should fill a led symbol but when it is `0` it should be transparent or as an alternative: black. When it is transparent it doesn't overwrite the filling and thus keeps the icon filled white.

When I want to fill the led symbol with black I'll need another transparency color. I tried it with `TFT_ORANGE` at `fillSprite` and `pushSprite` but then it draws a black box with the white led on top of a blue colored bar with has been drawn already by the `gfx`. This is instead of a transparent box.

Any ideas about this?

Bodmer, 2019-08-26:
X bitmap pixels are represented as 1 bit each so you can only have 2 colours (a bit can be 1 or 0).

The tft.drawXBitmap(...) function lets you specify only a foreground colour (only pixels set to 1 in the bitmap are drawn), or the foreground and background.  If you set both foreground and background to be the same colour then you can erase a plotted image.  See [this example](tft.drawXBitmap() which is commented and see how it works.

If you need more help then post a complete sketch that I can run and describe what you want it to do and what is not working as you would expect by adding comments in the code.

Joennuh, 2019-09-06:
Well thank you for your help offer but I expect I'll find a solution by my self. I think I will use tiny 16 bit color icons for it to overcome the issue. This is since I use blue gradient titlebar behind it. The only other solution I can think of to make the titlebar a solid color. Drawing the monochrome icons with that color could dissolve the icon in the background so the icon can be redrawn with other colors.

In the meanwhile I was trying so some stuff on a ST7735S connected to a Wemos D1 mini and the issue where this topic is all about can be reproduces on it. Also transparancy works fine if you set `gfx.setSwapBytes(false)` before `gfx.pushSprite(xpos,ypos,TRANSPARANCY_COLOR)` and after that `gfx.setSwapBytes(true)` again.

commit d3210a7e (2019-10-07): Fix #417 and add for 135x240 display

Setup 135 is for TTGO T Display (ESP32)
Plus minor tweaks to avoid some warnings
closed 2019-10-07 by Bodmer