Bodmer/TFT_eSPI #418: A few questions about displaying variables and text

pierpe83, 2019-08-27:
Hello.
Could you explain to me how the tft.drawFloat function works? What are the parameters of this function responsible for, in particular the "dp" parameter.
The second thing is to display the Float variable and the text to follow.
I would like to display the variable from the sensor and the text after it:
400.0 ppm
The variable has a different length and hence the problem with the position of the text after the variable. How to do it?
The third issue is refreshing the display. How to refresh the display without flashing?
Please help.
Greetings.

Bodmer, 2019-08-27:
Most libraries do not make this easy, but TFT_eSPI lets you define where the text datum is, so you can draw text and numbers that are, for example, left, centre or right justfiied. The setTestPAdding() function also allows old text to be obliterated with a fixed width background.

This is a simple example:

```
/*
Demostrates floating point number and string alignment

Normally strings are printed relative to the top left corner but this can be
changed with the setTextDatum() function. The library has #defines for:

TL_DATUM = Top left
TC_DATUM = Top centre
TR_DATUM = Top right
ML_DATUM = Middle left
MC_DATUM = Middle centre
MR_DATUM = Middle right
BL_DATUM = Bottom left
BC_DATUM = Bottom centre
BR_DATUM = Bottom right

 
 Needs font 4

 Make sure all the display driver and pin comnenctions are correct by
 editting the User_Setup.h file in the TFT_eSPI library folder.

 #########################################################################
 ###### DON'T FORGET TO UPDATE THE User_Setup.h FILE IN THE LIBRARY ######
 #########################################################################
 */

#include <TFT_eSPI.h> // Hardware-specific library

TFT_eSPI tft = TFT_eSPI();       // Invoke custom library

void setup(void) {
  Serial.begin(115200);
  tft.init();
  tft.setRotation(1);
  tft.fillScreen(TFT_BLACK);
  tft.setTextColor(TFT_WHITE, TFT_BLUE);
}

void loop() {

  // Find centre of screen
  uint16_t x = tft.width()/2;
  uint16_t y = tft.height()/2/

  // Set datum to Middle Right
  tft.setTextDatum(MR_DATUM);

  // Set the padding to the maximum width that the digits could occupy in font 4
  // This ensures small numbers obliterate large ones on the screen
  tft.setTextPadding( tft.textWidth("-88.88", 4) );

  // Creat a random signed floating point number in range -15 to +15
  float fpn = random(91)/3.0 - 15.0;

  // Draw a floating point number with 2 decimal places with right datum in font 4
  tft.drawFloat( fpn, 2, x, y, 4);

  // Reset text padding to 0 otherwise all future rendered strings will use it!
  tft.setTextPadding(0);
  
  // Set datum to Middle Left
  tft.setTextDatum(ML_DATUM);

  // Draw text with left datum in font 4
  tft.drawString(" Units", x, y, 4);

  delay(1000);
}
```

pierpe83, 2019-08-28:
Thank you very much for your reply.

Now I have a nice refresh of the screen without flashing effect.

There is also the issue of displaying the variable with its unit.
I want to display a numerical variable and its unit after it, one unit apart, e.g. "400 ppm". The problem is that the numeric variable has a different length and I do not know where to put the text with the unit "ppm" at a given moment. Is there any option to automatically place text after the interchangeable at a fixed distance or do I have to calculate the length of the variable and on this basis place the text after the variable?

Bodmer, 2019-08-28:
The example does what you ask,  it uses right justified numbers and left justified units with a single datum.  Study the example and experiment.

There is a function drawNumber(value, x, y, font) that you can use for integer values.

closed 2019-08-28 by Bodmer