Bodmer/TFT_eSPI #421: fillScreen() doesn't work properly when used for sprites

robcazzaro, 2019-09-02:
I'm building a Tetris clock using one of these boards with integrated IPS LCD 240x135 resolution https://github.com/Xinyuan-LilyGO/TTGO-T-Display. I'm using the latest version of your library with the setup.h file from the manufacturer. Everything works well and performs very fast.

To avoid flickering, I'm using a double buffering strategy using a sprite as big as the screen (ESP32 has enough memory). But it looks as if fillScreen() doesn't work as expected for sprites, filling only a portion of the screen (I suspect it's reversing the x and y dimensions of the screen). I can easily work around the problem using fillRect() with a rectangle as big as the screen (which is what fillScreen() does anyway behind the scenes), but I thought you'd want to know, since it took me a few minutes to figure out why the Tetris animation was leaving behind trails on a portion of the screen (in my code, the sprite and screen backgrounds are the same color). Changing the sprite rotation doesn't make any difference

Please see code example below. It draws a test string on the screen, (rotated to portrait), creates a sprite as big as the screen, fills the sprite with a background color, and draws the same string (reversed), pushing the sprite on the screen to display with no flickering

`#include <TFT_eSPI.h>
#include <SPI.h>

#ifndef TFT_DISPOFF
#define TFT_DISPOFF 0x28
#endif

#ifndef TFT_SLPIN
#define TFT_SLPIN   0x10
#endif

#define TFT_MOSI            19
#define TFT_SCLK            18
#define TFT_CS              5
#define TFT_DC              16
#define TFT_RST             23

#define TFT_BL          4  // Display backlight control pin

TFT_eSPI tft = TFT_eSPI(135, 240); // Invoke custom library

TFT_eSprite img = TFT_eSprite(&tft); 

void setup() {
    tft.init();
    tft.setRotation(1);
    tft.fillScreen(TFT_RED);
    tft.setTextSize(3);
    tft.setTextColor(TFT_WHITE);
    tft.setSwapBytes(true);

    tft.setCursor(20, 20);
    tft.print("0123456789");
    

    img.setColorDepth(8);
    img.createSprite(240, 135);
    //img.setRotation(0);   // Different rotation settings make no difference, it looks as if the sprite rotation is always the same as the screen
    img.setTextSize(3);
    img.setTextColor(TFT_YELLOW, TFT_BLUE);
    img.fillScreen(TFT_BLUE);     // doesn't properly fill the screen
    //img.fillRect(0,0,240,135, TFT_BLUE);    // This works
    img.setCursor(20, 20);
    img.print("9876543210");

    delay(2000);
    img.pushSprite(0, 0);
}

void loop() {
}`

Bodmer, 2019-09-03:
fillScreen() is for the TFT screen and uses the screen x,y pixel count. Use img.fillSprite(colour) for sprites. See example [here.](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Sprite/Sprite_drawPixel/Sprite_drawPixel.ino#L62)

Post back if that causes an issue.

robcazzaro, 2019-09-03:
Sorry, I completely missed fillSprite(), my bad. Indeed fillSprite() works perfectly for my case. Closing this issue with a big thank for a great library!

closed 2019-09-03 by robcazzaro