Bodmer/TFT_eSPI #422: TTOG - T ESP32 Rotate Display

medpixman, 2019-09-04:
This is a fantastic and comprehensive library! Thank-you so very much for all of your hard work!

Is there a command wiki? For example, how do I rotate the text display from Portrait to Landscape?
Using this tft.setRotation(1)? What values are accepted?

I'm using LILYGO® TTGO T-Display ESP32 CP2104 WiFi bluetooth Module 1.14 Inch LCD Development Board For Arduino from Banggood:

ESPRESSIF-ESP32 240MHz Xtensa® dual-core 32-bit LX6 microprocessor
IPS ST7789V 1.14 Inch

--

Thanks

Bodmer, 2019-09-04:
Have a look at this [tutorial](https://learn.adafruit.com/adafruit-gfx-graphics-library/rotating-the-display).
The Adafruit_GFX library has the same basic functions.

A command summary is on my TO DO list. I hope to find time to write a guide in October when I have some free time.

medpixman, 2019-09-04:
Thank-you SO MUCH for responding, and so quickly too!

I really appreciated it!

-- 

*James G. Smirniotopoulos, MD*
*Chief Editor, MedPix*®

*Special Volunteer, National Library of Medicine,
NIHWebsite: https://medpix.nlm.nih.gov/home
<https://medpix.nlm.nih.gov/home>Former Professor and Chair of Radiology
(retired)Uniformed Services University of the Health Sciences*

Google Voice 530 MEDPIX1 (530) 633-7491 [No Solicitations]
*YouTube Channel - http://www.youtube.com/user/TheMedpixman
<http://www.youtube.com/user/TheMedpixman>*

*"Science and the pursuit of knowledge have no country, no religion, and no
political affiliation" - Richard Feynman*

*"Living Rich in Spirit, with Family and Friends, is more important than
any material wealth." - **Δημήτριος Σμυρνιωτοπουλος*

"Success is *not* measured by who gets credit. ...
Success is measured by *what gets done*."
* - Connie Morella, U.S. Congresswoman*

"Do not plan to Destroy, What you are not also able to Create"
*- Pierre LaFontaine*

"If you never venture beyond what you know
... You've spawned your own limitations."

*- Sumner Redstone*

“A man travels the world in search of what he needs and returns home to
find it.”
* - George A. Moore 1852-1933*

"All science is either physics or stamp collecting."
*- Ernest Rutherford 1871–1937*

"Curiosity, the overwhelming desire to know, is not characteristic of dead
matter."
*- Isaac Asimov 1920-1992*

KianYeo, 2019-09-08:
Hi medpixman,

I noticed that I have the similar setup as you. I am using the ESP32 with a ST7789V (1.54"). However I am not getting the library to work. There is nothing displaying on the screen.

May I ask if you are using hardware SPI (MOSI - GPIO23 and SCLK - GPIO18) or you can assign any pins for your SPI? Also, how are you connecting your DC, CS and RST pin?

Thanks in advance!

> This is a fantastic and comprehensive library! Thank-you so very much for all of your hard work!
>
> Is there a command wiki? For example, how do I rotate the text display from Portrait to Landscape? [...]

medpixman, 2019-09-08:
I'm using this ESP32 240MHz Xtensa® dual-core 32-bit LX6 microprocessor
with IPS ST7789V 1.14 Inch from BangGood:
https://www.banggood.com/LILYGO-TTGO-T-Display-ESP32-CP2104-WiFi-bluetooth-Module-1_14-Inch-LCD-Development-Board-For-Arduino-p-1522925.html?rmmds=myorder&cur_warehouse=USA

Here the pinout from their website:
[image: image.png]
I'm using this sketch:

 /*

  This is a fantastic and comprehensive library! Thank-you so very much for
all of your hard work!

  Is there a command wiki? For example, how do I rotate the text display
from Portrait to Landscape?
  Using this tft.setRotation(1)? What values are accepted?

  I'm using LILYGOÂ® TTGO T-Display ESP32 CP2104 WiFi bluetooth Module 1.14
Inch LCD Development Board For Arduino from Banggood:

  ESPRESSIF-ESP32 240MHz XtensaÂ® dual-core 32-bit LX6 microprocessor
  IPSÂ ST7789V 1.14 Inch

*/

#include <TFT_eSPI.h>
#include <SPI.h>
#include "WiFi.h"
#include <Wire.h>
#include <Button2.h>
#include "esp_adc_cal.h"
#include "bmp.h"

#ifndef TFT_DISPOFF
#define TFT_DISPOFF 0x28
#endif

#ifndef TFT_SLPIN
#define TFT_SLPIN   0x10
#endif

#define TFT_MOSI            19
#define TFT_SCLK            18
#define TFT_CS              5
#define TFT_DC              16
#define TFT_RST             23

#define TFT_BL          4  // Display backlight control pin
#define ADC_EN          14
#define ADC_PIN         34
#define BUTTON_1        35
#define BUTTON_2        0

TFT_eSPI tft = TFT_eSPI(135, 240); // Invoke custom library
Button2 btn1(BUTTON_1);
Button2 btn2(BUTTON_2);

char buff[512];
int vref = 1100;
int btnCick = false;

//! Long time delay, it is recommended to use shallow sleep, which can
effectively reduce the current consumption
void espDelay(int ms)
{
  esp_sleep_enable_timer_wakeup(ms * 1000);
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
  esp_light_sleep_start();
}

[... 214 lines cut ...]
*"Living Rich in Spirit, with Family and Friends, is more important than
any material wealth." - **Δημήτριος Σμυρνιωτοπουλος*

"Success is *not* measured by who gets credit. ...
Success is measured by *what gets done*."
* - Connie Morella, U.S. Congresswoman*

"Do not plan to Destroy, What you are not also able to Create"
*- Pierre LaFontaine*

"If you never venture beyond what you know
... You've spawned your own limitations."

*- Sumner Redstone*

“A man travels the world in search of what he needs and returns home to
find it.”
* - George A. Moore 1852-1933*

"All science is either physics or stamp collecting."
*- Ernest Rutherford 1871–1937*

"Curiosity, the overwhelming desire to know, is not characteristic of dead
matter."
*- Isaac Asimov 1920-1992*

KianYeo, 2019-09-11:
Thanks @medpixman . It was really helpful!

closed 2019-10-03 by Bodmer