Bodmer/TFT_eSPI #424: setAddrWindow() troubles

themakerbro, 2019-09-08:
Hi! I suspect updates to the Adafruit library may need addressing(?). In issue #309 you helped me solve the skewing issue by using different parameters in the setAddrWindow() function. I tried uploading the updated example code and I see a similar skew than I did originally:

[image]

Moreover if I try different parameters on the function (e.g., `setAddrWindow(20,10,128,128);`), the image becomes garbled: 

[image]

Any suggestion are much appreciated!

Bodmer, 2019-09-14:
I will investigate, it may be that the library example needs to be updated.

Bodmer, 2019-10-03:
Now fixed in the Github copy of the example. Thanks for reporting the problem.

commit 9a84fb15 (2019-10-03): Fix #424
closed 2019-10-03 by Bodmer