Bodmer/TFT_eSPI #426: getTouch Causes crash

justcallmekoko, 2019-09-15:
I have part of your vertical scrolling code from your terminal example running as an rx_callback function for my ESP32 when it sniffs for packets.

While the sniffer sniffs packets, it displays them on the screen using your vertical scrolling code.

This is all happening in the rx_callback function so it runs parallel to the main loop.

The problem comes when I run getTouch in the main loop. I don't know if this is causing some sort of race condition, but if getTouch in loop runs at the same as the tft.print statement in the rx_callback, it causes the device to crash and reboot. If I comment out getTouch, it runs fine. If I comment out all of the tft.print statements but keep getTouch, it runs fine. I just can't have both.

Bodmer, 2019-09-16:
There are a number of reasons why an ESP32 will crash, the usual one is a watchdog timeout, in that case adding yield() or delay(0) at an appropriate point fixxes it. Other causes include high use of RAM leaving insufficient free and and running off the end of an array.

Please provide a simplified sketch that I can run and determine the root cause.

justcallmekoko, 2019-09-16:
I have pasted a simplified version of the code like you asked. It replicates the same issue perfectly. The crash you described is a watchdog timeout. I could not figure out where exactly to place the delay or yield.

I am also including the exact crash message:
`RSSI: -83 Ch: 2 Client: 68:e7:c2:66:da:b3E (8571) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (8571) task_wdt:  - IDLE0 (CPU 0)
E (8571) task_wdt: Tasks currently running:
E (8571) task_wdt: CPU 0: wifi
E (8571) task_wdt: CPU 1: loopTask
E (8571) task_wdt: Aborting.
abort() was called at PC 0x400dc277 on core 0

Backtrace: 0x4008c50c:0x3ffbe290 0x4008c73d:0x3ffbe2b0 0x400dc277:0x3ffbe2d0 0x4008165d:0x3ffbe2f0 0x400d2585:0x00000000

Rebooting...`

Here is the example code

`#include <TFT_eSPI.h>
#include "esp_wifi.h"
#include "esp_wifi_types.h"

#define SCREEN_WIDTH 240
#define SCREEN_HEIGHT 320
#define TEXT_HEIGHT 16 // Height of text to be printed and scrolled
#define BOT_FIXED_AREA 0 // Number of lines in bottom fixed area (lines counted from bottom of screen)
#define TOP_FIXED_AREA 16 // Number of lines in top fixed area (lines counted from top of screen)
#define YMAX 320 // Bottom of screen area
#define minimum(a,b)     (((a) < (b)) ? (a) : (b))

TFT_eSPI tft = TFT_eSPI();

const wifi_promiscuous_filter_t filt = {.filter_mask=WIFI_PROMIS_FILTER_MASK_MGMT | WIFI_PROMIS_FILTER_MASK_DATA};
uint32_t initTime = 0;
uint32_t currentTime = 0;
int set_channel = 1;

uint16_t yDraw = YMAX - BOT_FIXED_AREA - TEXT_HEIGHT;

// Keep track of the drawing x coordinate
uint16_t xPos = 0;

// The initial y coordinate of the top of the scrolling area
uint16_t yStart = TOP_FIXED_AREA;
// yArea must be a integral multiple of TEXT_HEIGHT
uint16_t yArea = YMAX - TOP_FIXED_AREA - BOT_FIXED_AREA;

// We have to blank the top line each time the display is scrolled, but this takes up to 13 milliseconds
// for a full width line, meanwhile the serial buffer may be filling... and overflowing
// We can speed up scrolling of short text lines by just blanking the character we drew
int blank[19]; // We keep all the strings pixel lengths to optimise the speed of the top line blanking

typedef struct
    {
      int16_t fctl;
      int16_t duration;
      uint8_t da;
      uint8_t sa;
      uint8_t bssid;
      int16_t seqctl;
      unsigned char payload[];
    } __attribute__((packed)) WifiMgmtHdr;
    
    typedef struct {
      WifiMgmtHdr hdr;
      uint8_t payload[0];
    } wifi_ieee80211_packet_t;
    

void clearScreen()
{
[... 161 lines cut ...]
  esp_wifi_init(&cfg);
  esp_wifi_set_storage(WIFI_STORAGE_RAM);
  esp_wifi_set_mode(WIFI_MODE_NULL);
  esp_wifi_set_promiscuous(true);
  esp_wifi_set_promiscuous_filter(&filt);
  esp_wifi_set_promiscuous_rx_cb(&sniffer_callback);
  esp_wifi_set_channel(set_channel, WIFI_SECOND_CHAN_NONE);
  initTime = millis();
}

void loop() {
  
  checkTouch();
  
  currentTime = millis();
  if (currentTime - initTime >= 1000)
  {
    initTime = millis();
    channelHop();
  }

  delay(1);

}
`

Bodmer, 2019-09-17:
OK. I can see the problem. Using the callback is effectively like and interrupt and if the callback includes SPI transactions (eg writes to TFT), then these might occur during SPI exchanges with the touch controller, so things get messed up because the chip select lines are software controlled.

You need to ensure the TFT and touch controller commands do not get interrupted by the callback. This can be achieved by leaving TFT display and touch controller commands out of the callback function. You would then need to queue up the text in a buffer for display on the TFT within the main loop().

closed 2019-09-17 by Bodmer