Bodmer/TFT_eSPI #427: Another 3.5" RPI Display

notascooby, 2019-09-16:
First of all - Great library.
Using ESP32, & 3.5" RPI Display ILI9486 (I think)
I have been using your library with 3.5" RPI displays for some time and all has been great, however I just bought a few more displays that are different to the normal displays sent. (Advert did show the normal type (A) though).
I was going to send them back but thought that the ability to use yet another 3.5" display would be useful for me and the community at large.
These new displays do work with 'Inversion On' however the R & G colours are switched.
I have tried TFT_BGR 1  &  TFT_RGB 1 but no difference. Guess these are for other displays or not used anymore. 
Also tried changing the RGB bit with command 0x36 but again not difference. I couldn't understand why this wouldn't work 
Also tried on my standard displays which are ILI9486, but neither of the above changed those displays either.
So I guess what I am asking is:
1 Am I missing something
2 If RGB - BGR switch not possible on 3.5" displays at present, could this be added (I would be willing to help test this if required)

I will try to send image of display later since for some reason I can't attach files at present!

Regards

notascooby, 2019-09-16:
Hi

Image of display as promised

Regards

[image]

Bodmer, 2019-10-07:
You could try changing [this file](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9486_Rotation.h) so that all instances of:
TFT_MAD_BGR 
read as:
TFT_MAD_RGB

This changes the colour order from BGR to RGB.
Then ensure you call setRotation() at least once, e.g. in setup() after the display has been initialised.

Let me know if that worked.

notascooby, 2019-10-15:
Thanks for response.
Unfortunately the display no longer works. Think the quality of these displays is not good, since touch was also very noisy.
If I get another one I will try your suggestion.
Thanks

closed 2019-11-02 by Bodmer