Bodmer/TFT_eSPI #428: loadFont causes Core 1 to panic on ESP32 version 1.0.3

Joennuh, 2019-09-16:
At ESP32 board package v1.0.3 Core 1 panic's at the `loadFont` instruction (to get smooth fonts on screen loaded from a font-file from SPIFFS).
At ESP board package v1.0.2 it doesn't panic (currently I downgraded to 1.0.2 to temporarily bypass the issue).

All fonts needed are available on SPIFFS and can be listed by my sketch. I think it should have something to do with the `loadFont` method.

This is the stacktrace and some serial output from my sketch:
```text
The Big Site Weatherstation V2
SW 0.1.0002
BOOT OK
SPIFFS initialisation...

SPIFFS available!

SPIFFS files found:
Listing directory: /
  /Final-Frontier-28.vlw             25287 bytes
  /NotoSansBold15.vlw                10766 bytes
  /NotoSansBold36.vlw                44169 bytes
  /NotoSansMonoSCB20.vlw             15382 bytes
  /OpenSans36.vlw                    88301 bytes

Checking font availability...

Fonts found OK.
Printing software version... Loading font... Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d744e  PS      : 0x00060130  A0      : 0x800d83f9  A1      : 0x3ffb1ee0  
A2      : 0x3ffc14cc  A3      : 0x0000ffff  A4      : 0x00000000  A5      : 0x001bfffc  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d744c  A9      : 0x3ffb1ec0  
A10     : 0xffffffff  A11     : 0xffff0000  A12     : 0xffffffff  A13     : 0x3f41d146  
A14     : 0x3ffb99bc  A15     : 0xff000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000  

Backtrace: 0x400d744e:0x3ffb1ee0 0x400d83f6:0x3ffb1f00 0x400d2f0e:0x3ffb1f50 0x400dcb37:0x3ffb1fb0 0x4008a479:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9564
ho 0 tail 12 room 4
load:0x40080400,len:6320
entry 0x400806a8

The Big Site Weatherstation V2
SW 0.1.0002
BOOT OK
```

And this is the decoded stacktrace with the ESP32 Exception Decoder tool (I have masked some personal information with asterisks):
```text
Decode stacktrace:

PC: 0x400d744e: TFT_eSPI::textWidth(char const*, unsigned char) at C:\Users\*****\Documents\Arduino\libraries\TFT_eSPI-master\TFT_eSPI.cpp line 2317
EXCVADDR: 0x00000000

Decoding stack results
0x400d744e: TFT_eSPI::textWidth(char const*, unsigned char) at C:\Users\*****\Documents\Arduino\libraries\TFT_eSPI-master\TFT_eSPI.cpp line 2317
0x400d83f6: fs::SPIFFSFS::SPIFFSFS(std::shared_ptr ) at c:\users\*****\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\ext/atomicity.h line 53
0x400d2f0e: setup() at C:\Arduino\*****/Weatherstation_version.ino line 729
0x4008a479: rtc_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/soc/esp32/rtc_init.c line 57
```
I'm quite new to Arduino and I don't know how this should be fixed. Al I now that for now downgrading to ESP32 version 1.0.2 temporarily fixes the issue.

Bodmer, 2019-09-17:
The SPIFFS partitioning or memory allocation in ESP32 1.0.3 has probably changed, thus data stored in SPIFFS under version 1.0.2 would then not be compatible with a sketch compiled under 1.0.3.

So I suspect this is not a TFT_eSPI issue.

Try loading 1.0.3 then select the SPIFFS size in the "Partition scheme.." menu option and then use the "ESP32 Sketch Data Upload" menu option. You should also compile and upload the sketch so everything loaded (compiled sketch and data in SPIFFS) are all loaded via the 1.0.3 version. This will ensure the sketch and the SPIFFS data format are correctly loaded that board package version.

Let me know if this is the solution.

Joennuh, 2019-09-17:
So when under version 1.0.3 of ESP32 I should re-upload the files? That is something I didn't do indeed. I will try this when I find the time to do it.

Bodmer, 2019-09-17:
Yes, reload the files to SPIFFS and compile/upload sketch as well using version 1.0.3

Joennuh, 2019-09-18:
Re-uploading the SPIFFS data with ESP32 version 1.0.3 as well re-uploading the sketch with ESP32 version 1.0.3 indeed resolve the issue.

Bodmer, 2019-09-18:
Great, thanks for reporting back.

JuergenLeber, 2020-01-26:
Having the same problem currently with 1.0.4 but re-uploading SPIFFS and the sketch unfortunately doesn't solve the problem. Any ideas?

Bodmer, 2020-01-26:
If you have upgraded to 1.0.4 then make sure you have done the following:
1. Updated the Arduino IDE SPIFFS loader to the latest
2. Uploaded the files using the new SPIFFS loader (you may need to exit the IDE to get the new loader adopted)
3. Recompile the sketch under 1.0.4 and upload the sketch again
4. Make sure the SPIFFS size set during the file upload to the ESP32 is the exact same size allocation as for the new sketch upload.

This will ensure the loaded SPIFFS format/size is compatible with 1.0.4  *AND* the sketch is compatible with the new format/size as well.

If this does not solve the problem then I need an a simple complete example sketch that demonstrates the problem plus as many details of your exact setup as you can give me (including board, TFT, connections and Arduino IDE version). This will allow me to setup the same conditions.

Unfortunately the simplicity of SPIFFS means it does not handle error states well (for example if sketch is uploaded with different SPIFFS size to the compiled sketch.

Bodmer, 2020-01-26:
Make sure the files fit in SPIFFS as well. You may not get an error message from the IDE if they do not!

JuergenLeber, 2020-01-26:
It's just two files, 55kb together. I assume you mean the IDE SPIFFS tool from me_no_dev, right? I already tried the smooth font demos with two fonts and it's working flawlessly. Then I tinkered with ArduinoMenu4 and also this is working. Next step was I wanted smooth fonts there and just started with "combining" it so initialized SPIFFS (which worked fine) and then immediately at "loadfont" it crashes. Need to check now if I'm using the correct esptool.py or if one comes inside the IDE integration from me_no_dev. Let me tinker a little bit and if I'm unlucky I'll send you a small sketch. Thanks in advance!

JuergenLeber, 2020-02-03:
No luck so far. I'll provide you a small sketch to reproduce the next days.

JuergenLeber, 2020-02-11:
I just made an example by using an ArduinoMenu4 example with a 1.77" TFT and a rotary encoder on an ESP-WROOM-32 dev board. TFT_eSPI is configured as follows:

`#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15
#define TFT_DC    2
#define TFT_RST   4`

Display itself and encoder are working fine in different examples. Still don't get why it crashes here. It sometimes also doesn't crash immediately after loadFont. But still no clue what is happening or what I'm doing wrong. Here the example sketch:

`#include <Arduino.h>
#include <FS.h>
#include <SPI.h>
#include <TFT_eSPI.h>
#include <menu.h>
#include <menuIO/TFT_eSPIOut.h>
#include <streamFlow.h>
#include <ClickEncoder.h>
// Using this library: https://github.com/soligen2010/encoder.git
#include <menuIO/clickEncoderIn.h>
#include <menuIO/keyIn.h>
#include <menuIO/chainStream.h>
#include <menuIO/serialIO.h>
// For debugging the TFT_eSPIOut.h library
#include <menuIO/serialOut.h>
#include <menuIO/serialIn.h>

using namespace Menu;

// Declare pins for rotary encoder
#define encA 21
#define encB 19
#define encBtn 5
#define encSteps 4

// Setup TFT colors.  Probably stop using these and use the colors defined by ArduinoMenu
#define BACKCOLOR TFT_BLACK
#define TEXTCOLOR TFT_WHITE

#define AA_FONT_SMALL "NotoSansBold15"
#define AA_FONT_LARGE "NotoSansBold36"

int chooseField = 1;
int cutsMade = 0;
int numberOfCuts = 5;
int lengthOfCuts = 50; // Length in mm
int feedLength = 304;
int exitMenuOptions = 0; //Forces the menu to exit and cut the copper tape

// Declare the clickencoder
// Disable doubleclicks in setup makes the response faster.  See: https://github.com/soligen2010/encoder/issues/6
ClickEncoder clickEncoder = ClickEncoder(encA, encB, encBtn, encSteps);
ClickEncoderStream encStream(clickEncoder, 1);

// TFT gfx is what the ArduinoMenu TFT_eSPIOut.h is expecting
TFT_eSPI gfx = TFT_eSPI();

void feedInOut();         // Function to manually feed the tape without cutting
void runCuts();           // Main loop that feeds tape and calls servoCuts()
void IRAM_ATTR onTimer(); // Start the timer to read the clickEncoder every 1 ms

//////////////////////////////////////////////////////////
// Start ArduinoMenu
//////////////////////////////////////////////////////////

result doFeed()
{
  delay(500);
  exitMenuOptions = 2;
  return proceed;
[... 166 lines cut ...]
      lastMenuFrame = millis();
      nav.poll(); // Poll the input devices
    }
  }
}

void runCuts()
{
  exitMenuOptions = 0;   // Return to the menu
  delay(500);
  mainMenu.dirty = true; // Force the main menu to redraw itself
}

void feedInOut()
{
  exitMenuOptions = 0;   // Return to the menu
  delay(500);
  mainMenu.dirty = true; // Force the main menu to redraw itself
}

// ESP32 timer
void IRAM_ATTR onTimer()
{
  clickEncoder.service();
}`

Bodmer, 2020-02-15:
I don't have the time to load your example, load all the libraries you use and then setup the required hardware environment... so you will have to work out a debug approach by disabling functions and see what is causing the issue.  The interrupts may be causing the problem, also you use pin 19 for the encoder but that is the default SPI MISO pin, so I suggest you use another pin. Bear in mind when the TFT is being updated the TFT_eSPI library claims exclusive access to the GPIO and SPI registers by using the SPI transaction mutex so that may prevent access to GPIO by an interrupt handler unless it correctly handles the mutex flags. These suggestions are all guesses and I can't help further unless there is an identifed issue specific to the TFT_eSPI library.

closed 2020-02-15 by Bodmer

JuergenLeber, 2020-02-24:
Thank you for the hint, it seems to be a problem with the encoder library in connection with TFT_eSPI. Will do further research and open another issue. At least "loadFont" has nothing to do with my issue.

pvlvsk, 2020-06-07:
Hello @JuergenLeber, 

could you solve your issue? i have a very similar problem with TFT_eSPI and an encoder library with esp32 constantly crashing (ai-esp32-rotary-encoder) when i read the encoder and afterwards print to the tft screen.

Thank you.

JuergenLeber, 2020-06-07:
Hi @pvlvsk,

unfortunately not. As soon as an encoder library with interrupt support is used this lead to spontaneous crashes. So if you can poll the encoder everything is fine.

Regards,
Jürgen

Bodmer, 2020-06-07:
I have used that encoder library without any problems... try this simple example which runs fine on my ESP32 setup:
```
#include "AiEsp32RotaryEncoder.h"
#include "Arduino.h"

#include <TFT_eSPI.h> // Hardware-specific library
TFT_eSPI tft = TFT_eSPI();       // Invoke custom library

#define ROTARY_ENCODER_A_PIN 32
#define ROTARY_ENCODER_B_PIN 33
#define ROTARY_ENCODER_BUTTON_PIN 21
#define ROTARY_ENCODER_VCC_PIN -1

AiEsp32RotaryEncoder rotaryEncoder = AiEsp32RotaryEncoder(ROTARY_ENCODER_A_PIN, ROTARY_ENCODER_B_PIN, ROTARY_ENCODER_BUTTON_PIN, ROTARY_ENCODER_VCC_PIN);

void rotary_loop() {
	if (rotaryEncoder.currentButtonState() == BUT_RELEASED) {
    Serial.println("Button released");
	}

	int16_t encoderDelta = rotaryEncoder.encoderChanged();
	
	if (encoderDelta == 0) return;
	
	if (encoderDelta!=0) {
		int16_t encoderValue = rotaryEncoder.readEncoder();
		Serial.print("Value: ");
		Serial.println(encoderValue);
	}
}

void setup() {
	Serial.begin(115200);
  tft.begin();
  tft.fillScreen(TFT_GREEN);

	rotaryEncoder.begin();
	rotaryEncoder.setup([]{rotaryEncoder.readEncoder_ISR();});
	rotaryEncoder.setBoundaries(0, 1000, true);
}

void loop() {
	rotary_loop();
  tft.fillScreen(random(0x10000));
  delay(100);
}
```

andrej-pavlov, 2020-06-10:
Hello @Bodmer,

indeed, this simple example is working for me without problems too. I haven't figured it out, but i think the problem / incompatibility comes out, when i start to use loadFont or sprites together with the encoder library...

Thank you