Bodmer/TFT_eSPI #429: LoRa stops working when running ESP32 DualCore with ILI9488 display

FLYINGSC0T, 2019-09-17:
Hi Bodmer,

Thanks again for the great work you do !!

My basic setup:
ESP32 NodeMCU
ILI9488 TFT
LoRa module RFM95W

TFT_eSPI GPIO's used:
#define TFT_CLK   5
#define TFT_MISO 19 // Touch T_D0 connected but not TFT SDO on ILI9488
#define TFT_MOSI 27
#define TFT_CS   15
#define TFT_DC    2
#define TFT_RST   4
#define TOUCH_CS  0

LoRa GPIO's used:
#define SCK_GPIO5       5
#define MISO_GPIO19    19
#define MOSI_GPIO27    27
#define SS_GPIO18      18
#define RST_GPIO14     14
#define DI0_GPIO26     26
#define LoRa_BAND      868E6

Everything works OK when running on one core (core 1) - with both ILI9341 and ILI9488 TFT displays.
Everything works OK when running on both cores - LoRa running on core 1 and TFT running on core 0 with ILI9341 display ONLY.
Have tested the above configurations overnight and all LoRa packets were received successfully without the LoRa module stopping.

Now the problem...
Does not work running on both cores - LoRa running on core 1 and TFT running on core 0 with the ILI9488 display. Have also tried swapping LoRa to core 0 and TFT to core 1 - still the same.
LoRa stops working intermittently and needs to be re-started. I have included code to restart the LoRa module after 1 minute if no packets are received. It restarts and will immediately capture the incoming packets. But the same thing happens again and again.

Here's the output from the serial monitor:
17:18:15.280 -> LoRa started OK !!
17:18:15.314 -> Initialized OK
17:18:15.314 -> LoRa running on core: 1
17:18:15.823 -> TFT running on core: 0
17:18:16.026 -> TFT Count: 1
17:18:16.333 -> Packet count: 1
17:18:16.333 -> Received packet '$GAFSOL,6,38.538250,12.835301,3093,135,125,F51BEDE6' with RSSI -41
17:18:17.245 -> TFT Count: 2
17:18:18.495 -> TFT Count: 3
17:18:18.495 -> Packet count: 1
17:18:18.495 -> Received packet '$,,P,44,⸮⸮⸮+66,⸮⸮⸮*⸮⸮+22,⸮⸮⸮+44,⸮⸮' with RSSI -41
17:18:19.717 -> TFT Count: 4
17:18:20.938 -> TFT Count: 5
17:18:22.153 -> TFT Count: 6
17:18:23.405 -> TFT Count: 7
17:18:24.625 -> TFT Count: 8
17:18:25.844 -> TFT Count: 9
17:18:27.060 -> TFT Count: 10
17:18:28.309 -> TFT Count: 11
17:18:29.526 -> TFT Count: 12
17:18:30.737 -> TFT Count: 13
17:18:31.984 -> TFT Count: 14
17:18:33.197 -> TFT Count: 15
17:18:34.444 -> TFT Count: 16
17:18:35.664 -> TFT Count: 17
17:18:36.882 -> TFT Count: 18
17:18:38.101 -> TFT Count: 19
17:18:39.352 -> TFT Count: 20
17:18:40.570 -> TFT Count: 21
17:18:41.785 -> TFT Count: 22
17:18:43.035 -> TFT Count: 23
17:18:44.250 -> TFT Count: 24
17:18:45.469 -> TFT Count: 25
17:18:46.715 -> TFT Count: 26
17:18:47.931 -> TFT Count: 27
17:18:49.152 -> TFT Count: 28
17:18:50.404 -> TFT Count: 29
17:18:51.620 -> TFT Count: 30
17:18:52.835 -> TFT Count: 31
17:18:54.084 -> TFT Count: 32
17:18:55.301 -> TFT Count: 33
17:18:56.518 -> TFT Count: 34
17:18:57.763 -> TFT Count: 35
17:18:58.981 -> TFT Count: 36
17:19:00.199 -> TFT Count: 37
17:19:01.453 -> TFT Count: 38
17:19:02.667 -> TFT Count: 39
17:19:03.914 -> TFT Count: 40
17:19:05.128 -> TFT Count: 41
17:19:06.344 -> TFT Count: 42
17:19:07.563 -> TFT Count: 43
17:19:08.814 -> TFT Count: 44
17:19:10.029 -> TFT Count: 45
17:19:11.246 -> TFT Count: 46
17:19:12.492 -> TFT Count: 47
17:19:13.708 -> TFT Count: 48
17:19:14.960 -> TFT Count: 49
17:19:16.177 -> TFT Count: 50
17:19:17.394 -> TFT Count: 51
17:19:18.642 -> TFT Count: 52
17:19:19.047 -> LoRa re-started !!
17:19:19.856 -> TFT Count: 53
17:19:20.059 -> Packet count: 2
17:19:20.059 -> Received packet '$GAFSOL,5,52.314190,12.847804,3085,45,120,99F06271' with RSSI -38
17:19:21.073 -> TFT Count: 54
17:19:22.081 -> Packet count: 3
17:19:22.081 -> Received packet '$GAFSOL,7,38.525738,-0.927649,3085,225,130,E1006751' with RSSI -39
17:19:22.317 -> TFT Count: 55
17:19:23.564 -> TFT Count: 56
17:19:24.778 -> TFT Count: 57
17:19:25.996 -> TFT Count: 58
17:19:27.247 -> TFT Count: 59
17:19:28.466 -> TFT Count: 60
17:19:29.280 -> Packet count: 4
17:19:29.280 -> Received packet '$GAFSOL,8,52.315441,-0.928899,3089,315,135,927089B3' with RSSI -40
17:19:29.719 -> TFT Count: 61
17:19:30.942 -> TFT Count: 62
17:19:32.162 -> TFT Count: 63
17:19:33.410 -> TFT Count: 64
17:19:33.410 -> Packet count: 4
17:19:33.410 -> Received packet '$,⸮P|~|⸮|~|@⸮@⸮⸮|~|⸮|~|' with RSSI -40
17:19:34.627 -> TFT Count: 65
17:19:35.846 -> TFT Count: 66
17:19:37.060 -> TFT Count: 67
17:19:38.305 -> TFT Count: 68
17:19:39.523 -> TFT Count: 69
17:19:40.743 -> TFT Count: 70
17:19:41.989 -> TFT Count: 71
17:19:43.202 -> TFT Count: 72
17:19:44.416 -> TFT Count: 73
17:19:45.666 -> TFT Count: 74
17:19:46.882 -> TFT Count: 75
17:19:48.099 -> TFT Count: 76
17:19:49.348 -> TFT Count: 77
17:19:50.560 -> TFT Count: 78
17:19:51.774 -> TFT Count: 79
17:19:53.023 -> TFT Count: 80
17:19:54.238 -> TFT Count: 81
17:19:55.462 -> TFT Count: 82
17:19:56.682 -> TFT Count: 83
17:19:57.933 -> TFT Count: 84
17:19:59.146 -> TFT Count: 85
17:20:00.363 -> TFT Count: 86
17:20:01.584 -> TFT Count: 87
17:20:02.835 -> TFT Count: 88
17:20:04.057 -> TFT Count: 89
17:20:05.276 -> TFT Count: 90
17:20:06.493 -> TFT Count: 91
17:20:07.710 -> TFT Count: 92
17:20:08.962 -> TFT Count: 93
17:20:10.179 -> TFT Count: 94
17:20:11.393 -> TFT Count: 95
17:20:12.643 -> TFT Count: 96
17:20:13.856 -> TFT Count: 97
17:20:15.073 -> TFT Count: 98
17:20:16.321 -> TFT Count: 99
17:20:17.535 -> TFT Count: 100
17:20:18.749 -> TFT Count: 101
17:20:19.998 -> TFT Count: 102
17:20:21.210 -> TFT Count: 103
17:20:22.425 -> TFT Count: 104
17:20:23.676 -> TFT Count: 105
17:20:24.895 -> TFT Count: 106
17:20:26.113 -> TFT Count: 107
17:20:27.327 -> TFT Count: 108
17:20:28.575 -> TFT Count: 109
17:20:29.789 -> TFT Count: 110
17:20:31.005 -> TFT Count: 111
17:20:32.253 -> TFT Count: 112
17:20:33.468 -> TFT Count: 113
17:20:33.906 -> LoRa re-started !!
17:20:34.684 -> TFT Count: 114
17:20:34.887 -> Packet count: 5
17:20:34.887 -> Received packet '$GAFSOL,1,52.329205,5.959611,3078,360,100,E65B6CDF' with RSSI -41
17:20:35.931 -> TFT Count: 115
17:20:37.146 -> TFT Count: 116
17:20:37.953 -> Packet count: 6
17:20:37.953 -> Received packet '$GAFSOL,4,45.419964,-0.942649,3078,270,115,1B63044B' with RSSI -39
17:20:38.391 -> TFT Count: 117
17:20:39.604 -> TFT Count: 118
17:20:40.851 -> TFT Count: 119
17:20:42.097 -> TFT Count: 120
17:20:43.313 -> TFT Count: 121
17:20:44.560 -> TFT Count: 122
17:20:45.134 -> Packet count: 7
17:20:45.134 -> Received packet '$GAFSOL,5,52.330456,12.864058,3079,45,120,C2F92DF9' with RSSI -40
17:20:45.775 -> TFT Count: 123

All packets with "$GAFSOL" are correct but, for example, the received packet "$,⸮P|~|⸮|~|@⸮@⸮⸮|~|⸮|~|" is not what was transmitted - LoRa stops working.

So what's causing this issue when using the ILI9488 display as the ILI9341 works OK ?

BTW the TFT Counts are at one second intervals and are also output to the display along with some other information.

Thanks in advance anyone for any insight in solving this issue.

Bodmer, 2019-09-17:
I suspect that the problem is experienced with the ILI9488 because that display uses 8 bit commands and 18 bit SPI transactions for the pixel colours, whereas the ILI9341 uses 8 bit commands and 16 bit transactions for pixels.

This suggests that the LoRa library you are using does not support shared transactions on the SPI bus and maybe assumes it has exclusive use of the SPI bus and it is always configured for 16 bit exchanges.

This is just a guess as I do not know what software libraries you are using.  If the sketch uses callbacks then this can also cause problems because TFT transactions might get interrupted by SPI exchanges with the RFM95W. The ESP32 has 2 SPI ports so if the pins are spare you could configure the TFT_eSPI library to [use the other port](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L266-L269)

closed 2019-09-25 by Bodmer

FLYINGSC0T, 2019-09-25:
Hi Bodmer,

Here's an update as I've now got it working !!

This is what I've done:
Kept the same GPIO's as previously (LoRa default) as the LoRa RFM95W module has problems starting if other GPIO's are used.
Set up the two tasks to run using two threads and assigned them to both run on core 1.
Set priority of the TFT task to 1 and the LoRa task to 2 (with the same priority it doesn't work).

Currently the program has been running for almost 22 hours now and approaching 80000 packets received with no drop-out !!
With the new setup, packets are being received in less than a second - incredible !!

My conclusions:
It is imperative to use the default GPIO's for the LoRa module viz: 
GPIO5  -- SX1278's SCK
GPIO19 -- SX1278's MISO
GPIO27 -- SX1278's MOSI
GPIO18 -- SX1278's CS
GPIO14 -- SX1278's RESET
GPIO26 -- SX1278's IRQ (Interrupt Request)

Share the same SPI bus with the ILI9488 display otherwise it doesn't work (MISO is not connected on TFT as per your advice).
HSPI cannot be used with TFT_eSPI library as LoRa uses GPIO 14 for it's RESET - changing this pin then LoRa module will not start.

As I have found out from advice on the Espressif forum, it seems to be a threading problem.
I've also tested further to see if the same configuration works running on both cores but it doesn't.

I can now stop pulling my hair out - I'm almost bald with this issue !!

Regards.

Bodmer, 2019-09-25:
Great to hear you have solved the problem. I have not experimented with using threaded software.