Bodmer/TFT_eSPI #430: LILYGO® TTGO T-Display ESP32 CP2104 WiFi bluetooth Module Touch Screen Compiler problem

Bushedo, 2019-09-17:
Have been unable to get the touch screen to work with the Bodmer library although all other display code seems to work as intended.  Keep getting "class TFT_eSPI has no member named getTouch". Have a Touch_CS line specified in the setup file, and moved the touch.h and .cpp files into main directory, but no luck. Can you offer any suggestions/help?
Trying to run the example sketches, but none of touch screen items will compile. 
In User_Setup_Select: #include <User_Setups/TTGO_T_Display.h>
In User_Setup:
 #define ST7789_DRIVER 
#define TFT_SDA_READ
#define TFT_RGB_ORDER TFT_RGB
#define TFT_WIDTH  135  
#define TFT_HEIGHT 240
#define TFT_MOSI 19
#define TFT_SCLK 18
#define TFT_CS    5  // Chip select control pin
#define TFT_DC    16  // Data Command control pin
#define TFT_RST   2  // Reset pin (could connect to RST pin)
#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST
#define TFT_BL   14  // LED back-light (only for ST7789 with backlight control pin)
#define TOUCH_CS 21     // Chip select pin (T_CS) of touch screen (NOT SURE THIS IS CORRECT 
                                         PIN, BUT DON'T THINK COMPILER WOULD KNOW THE DIFFERENCE)

Bushedo, 2019-09-18:
Well, it has been suggested that this device doesn't have a touch screen.  Not sure what gave me the impression it did.  Wasted a lot of time.  That would explain why I couldn't identify a chip to manage the touch screen on the schematic.  Thanks anyway.

BirdAPI, 2019-09-21:
@Bushedo I think your confusion probably comes from their [pinout image](https://github.com/Xinyuan-LilyGO/TTGO-T-Display/raw/master/image/pinmap.jpg). The `touch` pins are actually referring to the ESP32's native capacitive touch sensors.

[Here is a random tutorial online about them](https://randomnerdtutorials.com/esp32-touch-pins-arduino-ide/)

Bushedo, 2019-09-21:
BirdAPI,
You may be right or I saw a post from someone with a different board that had a touch screen.  Thanks.

Bodmer, 2019-09-25:
The library only compiles in the touch support routines if TOUCH_CS is defined, make sure you have included it in the right setup file. Only one setup file should be called up.

As you found the T Display does not have a touch screen.

closed 2019-09-25 by Bodmer