Bodmer/TFT_eSPI #432: STM32F103 support

arnold-b, 2019-09-21:
Hello, 

I like this library. Its the only one which realy works. 
Is there any posibility that you could add STM support there. 

Thanks

Bodmer, 2019-09-25:
I will think on this further. At the moment I do not have enpugh spare time to make and test the changes. The adaption is further complicated because some capabilities, like SPIFFS are espressif specific.
I am hoping to find time to write a library user manual before making any significant changes to the library.

Bodmer, 2019-11-02:
It appears that [SEEED Studio have adapted the library](https://github.com/Seeed-Studio/Seeed_Arduino_LCD) in a way that could support STM processors.

closed 2019-11-02 by Bodmer