Bodmer/TFT_eSPI #436: problem with ESP32_SDcard_Jpg

haseoleonard, 2019-10-02:
I got some problem with the example can someone assists

> Arduino: 1.8.10 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 921600"
> 
> C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\haseo\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\haseo\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\haseo\OneDrive\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=4M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -ide-version=10810 -build-path C:\Users\haseo\AppData\Local\Temp\arduino_build_225908 -warnings=none -build-cache C:\Users\haseo\AppData\Local\Temp\arduino_cache_19162 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python.path=C:\Users\haseo\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.python-3.7.2-post1.path=C:\Users\haseo\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\haseo\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=C:\Users\haseo\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs.path=C:\Users\haseo\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=C:\Users\haseo\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -verbose C:\Users\haseo\OneDrive\Documents\Arduino\libraries\TFT_eSPI\examples\Generic\ESP32_SDcard_jpeg\ESP32_SDcard_jpeg.ino
> C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\haseo\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\haseo\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\haseo\OneDrive\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=4M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -ide-version=10810 -build-path C:\Users\haseo\AppData\Local\Temp\arduino_build_225908 -warnings=none -build-cache C:\Users\haseo\AppData\Local\Temp\arduino_cache_19162 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python.path=C:\Users\haseo\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.python-3.7.2-post1.path=C:\Users\haseo\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\haseo\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=C:\Users\haseo\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs.path=C:\Users\haseo\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=C:\Users\haseo\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -verbose C:\Users\haseo\OneDrive\Documents\Arduino\libraries\TFT_eSPI\examples\Generic\ESP32_SDcard_jpeg\ESP32_SDcard_jpeg.ino
> Using board 'd1_mini' from platform in folder: C:\Users\haseo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2
> Using core 'esp8266' from platform in folder: C:\Users\haseo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2
> Detecting libraries used...
> "C:\\Users\\haseo\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\haseo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/include" "-IC:\\Users\\haseo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/lwip2/include" "-IC:\\Users\\haseo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\haseo\\AppData\\Local\\Temp\\arduino_build_225908/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10810 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"" -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\haseo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\cores\\esp8266" "-IC:\\Users\\haseo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\variants\\d1_mini" "C:\\Users\\haseo\\AppData\\Local\\Temp\\arduino_build_225908\\sketch\\ESP32_SDcard_jpeg.ino.cpp" -o nul
> Alternatives for SPI.h: [SPI@1.0]
> ResolveLibrary(SPI.h)
>   -> candidates: [SPI@1.0]
> "C:\\Users\\haseo\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\haseo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/include" "-IC:\\Users\\haseo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/lwip2/include" "-IC:\\Users\\haseo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\haseo\\AppData\\Local\\Temp\\arduino_build_225908/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10810 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"" -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\haseo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\cores\\esp8266" "-IC:\\Users\\haseo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\variants\\d1_mini" "-IC:\\Users\\haseo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\libraries\\SPI" "C:\\Users\\haseo\\AppData\\Local\\Temp\\arduino_build_225908\\sketch\\ESP32_SDcard_jpeg.ino.cpp" -o nul
> Alternatives for SD.h: [SD@1.2.3 SD@2.0.0]
> ResolveLibrary(SD.h)
>   -> candidates: [SD@1.2.3 SD@2.0.0]
> "C:\\Users\\haseo\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\haseo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/include" "-IC:\\Users\\haseo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/lwip2/include" "-IC:\\Users\\haseo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\haseo\\AppData\\Local\\Temp\\arduino_build_225908/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10810 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"" -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\haseo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\cores\\esp8266" "-IC:\\Users\\haseo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\variants\\d1_mini" "-IC:\\Users\\haseo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\libraries\\SPI" "-IC:\\Users\\haseo\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\libraries\\SD\\src" "C:\\Users\\haseo\\AppData\\Local\\Temp\\ard [... cut ...]
> Using library SDFS at version 0.1.0 in folder: C:\Users\haseo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS 
> Using library ESP8266SdFat at version 1.0.16 in folder: C:\Users\haseo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266SdFat 
> Using library TFT_eSPI at version 1.4.16 in folder: C:\Users\haseo\OneDrive\Documents\Arduino\libraries\TFT_eSPI 
> Using library JPEGDecoder at version 1.8.0 in folder: C:\Users\haseo\OneDrive\Documents\Arduino\libraries\JPEGDecoder 
> exit status 1
> no matching function for call to 'SDClass::begin()'

Joennuh, 2019-10-02:
Posting your sketch or only the relevant parts of your sketch would be helpfull to determine the possible cause. Also: which libraries do you use on your sketch?

haseoleonard, 2019-10-02:
well actually I use the exact example within the TFT_eSPI lib along side with library for esp8266

Joennuh, 2019-10-02:
Sorry. Didn't notice that in the long text.
Maybe your issue has something to do with a recent update of the ESP8266 board definitions.

haseoleonard, 2019-10-02:
I'm just starting to learn using ESP8266 so it's kinda confusing at the moment

haseoleonard, 2019-10-02:
@Joennuh Would you mind trying out the example files and tell me if it worked ? thanks ~

Bodmer, 2019-10-03:
The example you are trying is for the ESP32 processor, not the ESP8266. Unfortunately the SD card libraries for these two processors are not compatible, hence the error messages.

closed 2019-10-07 by Bodmer