Bodmer/TFT_eSPI #438: Touch controller functions with parallel interface displays

momosh13, 2019-10-05:
Hi 
I love this library, the best library on internet.
Please, if possible, enable touch controller functions for parallel interface displays, I'm using ILI9488.
Thank you

Bodmer, 2019-10-06:
If the display board is fitted with a resistance based touch screen then this can be used by performing the modifications described here and the [fork of the Adafruit library](https://github.com/s60sc/Adafruit_TouchScreen):

momosh13, 2019-10-06:
The is ILI9488 with Capacitive Touch / controller is FT6206, I'm using this screen one year with Your library everything is working fine, but I'm trying to use [LittleVGL Library](https://littlevgl.com)
it's bast on TFT-eSPI and it's requires touch controller functions to be enabled.
With SPI connection displays everything works fine, but SPI is very slow, that's why I wanna use the parallel display.
Please help.
Thank you

Bodmer, 2019-10-07:
I am not familiar with the FT6206 so am unable to help, if it has an I2C interface then maybe [this library](https://github.com/adafruit/Adafruit_FT6206_Library) will work?

closed 2019-10-07 by Bodmer