Bodmer/TFT_eSPI #439: Doubt on how to connect to esp8266

beckmx, 2019-10-12:
I have a TFT from raspberry, I connected as indicated in the arduino forum https://forum.arduino.cc/index.php?topic=443787.105:

```
+5V =>Vin (5V)
GND => GND

MOSI => pin D7
SCK  => pin D5
DC  => pin D3
CS => pin D8
RST => pin D4
TP_CS => 3.3V (Touch screen disabled)
```
And I try to compile the example:

```
/*
Tests string alignment

Normally strings are printed relative to the top left corner but this can be
changed with the setTextDatum() function. The library has #defines for:

TL_DATUM = Top left
TC_DATUM = Top centre
TR_DATUM = Top right
ML_DATUM = Middle left
MC_DATUM = Middle centre
MR_DATUM = Middle right
BL_DATUM = Bottom left
BC_DATUM = Bottom centre
BR_DATUM = Bottom right

 
 Needs fonts 2, 4, 6, 7 and 8

 Make sure all the display driver and pin comnenctions are correct by
 editting the User_Setup.h file in the TFT_eSPI library folder.

 #########################################################################
 ###### DON'T FORGET TO UPDATE THE User_Setup.h FILE IN THE LIBRARY ######
 #########################################################################
 */

#include <TFT_eSPI.h> // Hardware-specific library
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI();       // Invoke custom library

unsigned long drawTime = 0;

void setup(void) {
  Serial.begin(115200);
  tft.init();
  tft.setRotation(1);
}

void loop() {

  tft.fillScreen(TFT_BLACK);
  
  for(byte datum = 0; datum < 9; datum++) {
    tft.setTextColor(TFT_WHITE, TFT_BLACK);
    
    tft.setTextDatum(datum);
    
    tft.drawNumber(88,160,120,8);
    tft.fillCircle(160,120,5,TFT_RED);
    
    tft.setTextDatum(MC_DATUM);
    
    tft.setTextColor(TFT_BLACK);
    tft.drawString("X",160,120,2);
    delay(1000);
    tft.fillScreen(TFT_BLACK);
  }

  tft.setTextColor(TFT_BLUE, TFT_BLACK);
  tft.drawCentreString("69",160,120,8);
  tft.fillCircle(160,120,5,TFT_YELLOW);
  
  tft.setTextDatum(MC_DATUM);
  
  tft.setTextColor(TFT_BLACK);
  tft.drawString("X",160,120,2);
  delay(1000);
  tft.fillScreen(TFT_BLACK);
  
  tft.setTextColor(TFT_RED, TFT_BLACK);
  tft.drawRightString("88",160,120,8);
  tft.fillCircle(160,120,5,TFT_YELLOW);
  
  tft.setTextDatum(MC_DATUM);
  
  tft.setTextColor(TFT_BLACK);
  tft.drawString("X",160,120,2);
  delay(1000);
  tft.fillScreen(TFT_BLACK);

  tft.setTextColor(TFT_WHITE, TFT_BLUE);

  tft.setTextDatum(MC_DATUM);

  //Test floating point drawing function
  float test = 67.125;
  tft.drawFloat(test, 4, 160, 180, 4);
  delay(1000);
  tft.fillScreen(TFT_BLACK);
  test = -0.555555;
  tft.drawFloat(test, 3, 160, 180, 4);
  delay(1000);
  tft.fillScreen(TFT_BLACK);
  test = 0.1;
  tft.drawFloat(test, 4, 160, 180, 4);
  delay(1000);
  tft.fillScreen(TFT_BLACK);
  test = 9999999;
  tft.drawFloat(test, 1, 160, 180, 4);
  delay(1000);
  
  tft.fillCircle(160,180,5,TFT_YELLOW);
  
  tft.setTextDatum(MC_DATUM);
  
  tft.setTextColor(TFT_BLACK);
  tft.drawString("X",160,180,2);
 
  delay(4000);
}

```
But I dont see anything in the screen, I was wondering if I have to specify the chipset of the lcd or it automatically knows what is the one used

Bodmer, 2019-10-14:
Edit the User_Setup_Select.h file to [comment out line 22](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L22) and [uncomment line 28](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L28).

Then make sure the correct pins are defined in the called up [setup file](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup5_RPi_ILI9486.h).

Typical connections are:

// Display SDO/MISO  to NodeMCU pin D6 (or leave disconnected if not reading TFT)
// Display LED       to NodeMCU pin VIN (or 5V, see below)
// Display SCK       to NodeMCU pin D5
// Display SDI/MOSI  to NodeMCU pin D7
// Display DC (RS/AO)to NodeMCU pin D3
// Display RESET     to NodeMCU pin D4 (or RST, see below)
// Display CS        to NodeMCU pin D8 (or GND, see below)
// Display GND       to NodeMCU pin GND (0V)
// Display VCC       to NodeMCU 5V or 3.3V

beckmx, 2019-10-14:
well, mm,  I still see the screen gray, I was reading the docs, I am using the wemos d1, with this pinout (https://hobbytronics.pk/wp-content/uploads/wemos-pinout.jpg) however I see that the nodemcu uses the same references:

```
12:02:33.337 -> [code]
12:02:33.337 -> TFT_eSPI ver = 1.4.5
12:02:33.337 -> Processor    = ESP8266
12:02:33.337 -> Frequency    = 80 MHz
12:02:33.337 -> Voltage      = %.2f V
12:02:33.337 -> Transactions = No 
12:02:33.337 -> Interface    = SPI 
12:02:33.337 -> SPI overlap  = No 
12:02:33.337 -> 
12:02:33.337 -> Display driver = 9486
12:02:33.373 -> Display width  = 320 
12:02:33.373 -> Display height = 480 
12:02:33.373 -> 
12:02:33.373 -> TFT_CS   = D8 (GPIO 15)
12:02:33.373 -> TFT_DC   = D3 (GPIO 0)
12:02:33.373 -> TFT_RST  = D4 (GPIO 2)
12:02:33.373 -> 
12:02:33.373 -> Font GLCD   loaded
12:02:33.373 -> Font 2      loaded
12:02:33.373 -> Font 4      loaded
12:02:33.373 -> Font 6      loaded
12:02:33.373 -> Font 7      loaded
12:02:33.373 -> Font 8      loaded
12:02:33.373 -> Smooth font enabled
12:02:33.373 -> 
12:02:33.373 -> Display SPI frequency = %.1f MHz 
12:02:33.373 -> [/code]

```

I was wondering if the overlap is needed as well?

beckmx, 2019-10-14:
does this applies to the 3.2 inch display? because from what I see in the docs, looks like the library is for the 480x320? or it does applies for the lower resolutions?

Bodmer, 2019-10-14:
The display must be [exactly like this one](https://github.com/Bodmer/TFT_eSPI/blob/master/Tools/RPi_TFT_Connections.png) and have a 320x480 size.

Old D1 boards had different pin naming so in that case use the number only for the pin.

beckmx, 2019-10-14:
well the display is not exactly that one, is [this one 3.2 inches ](https://es.aliexpress.com/item/32914941731.html?spm=a2g0o.productlist.0.0.77e75107nQfaBt&algo_pvid=496a9aa7-21af-4839-bb2b-e4ad97bdc967&algo_expid=496a9aa7-21af-4839-bb2b-e4ad97bdc967-12&btsid=6efde22a-df7d-4089-add0-b9beb9dc2797&ws_ab_test=searchweb0_0,searchweb201602_1,searchweb201603_52), so, both 3.2 and 3.5 are different right? from what I see in the pictures it has some additional chipsets

Bodmer, 2019-10-14:
Perhaps try the ILI9341 setup.
Unfortunately I do not have a display like yours so it will be difficult to advise further.

closed 2019-10-14 by Bodmer