Bodmer/TFT_eSPI #440: Wemos D1 R32 and Elegoo 2.8 TFT touch

feendrache, 2019-10-12:
Hey there,

i've got the same board you show ín the TFT_eSPI Description, made the wiring and got it to work without touch... now i tried to get the touch to work too... following the incstructions of the fork you mentioned but i'm not able to get it work... 
With your wiring the display shows everyting correct,
but the wiring of the touch won't work... 
If i add 
#define TOUCH_CS 39     // Chip select pin (T_CS) of touch screen 

to my settings for the TFT_eSPI
i get errors like this:
`C:\Users\Naddy\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::spi_begin_touch()':

C:\Users\Naddy\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:108:34: error: 'spi' was not declared in this scope

     if (locked) {locked = false; spi.beginTransaction(SPISettings(SPI_TOUCH_FREQUENCY, MSBFIRST, SPI_MODE0));}

                                  ^

C:\Users\Naddy\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::spi_end_touch()':

C:\Users\Naddy\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:124:54: error: 'spi' was not declared in this scope

     if(!inTransaction) {if (!locked) {locked = true; spi.endTransaction();}}

                                                      ^

In file included from C:\Users\Naddy\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:5360:0:

C:\Users\Naddy\Documents\Arduino\libraries\TFT_eSPI\Extensions/Touch.cpp: In member function 'uint8_t TFT_eSPI::getTouchRaw(uint16_t*, uint16_t*)':

C:\Users\Naddy\Documents\Arduino\libraries\TFT_eSPI\Extensions/Touch.cpp:23:3: error: 'spi' was not declared in this scope

   spi.transfer(0xd0);                    // Start new YP conversion

   ^

C:\Users\Naddy\Documents\Arduino\libraries\TFT_eSPI\Extensions/Touch.cpp: In member function 'uint16_t TFT_eSPI::getTouchRawZ()':

C:\Users\Naddy\Documents\Arduino\libraries\TFT_eSPI\Extensions/Touch.cpp:66:3: error: 'spi' was not declared in this scope

   spi.transfer(0xb0);               // Start new Z1 conversion`

can you help me or give me a hint what i'm doing wrong?

Bodmer, 2019-10-14:
The additional touch library is a stand alone library that I have not tried. Do not define TOUCH_CS in the setup as otherwise the compiler will try to use the SPI library which is not wanted.

feendrache, 2019-10-16:
Thank you... 

for now i will stop trying to get the parallel display working with the Wemos d1 R32 since my main goal in my current project is not getting the display to work.

I got some touch-results from the library but cannot get an x value and the other values are... mysterious at best...
For now i ordered an SPI Touch display ... came in today, will be tested tomorrow... maybe with time, when i've got more experience i will try again to get these two things to work together

With your instructions on the Wemos D1 R32 rewiring i got it working so far and display stuff... wich i really appreciate

closed 2019-10-16 by feendrache