Bodmer/TFT_eSPI #441: ILI9341 Touch screen no response

UmarDastgir, 2019-10-13:
Hello. I recently bought a ILI9341 touchscreen (2.8 inch) and interfaced it with an ESP8266. I was having some troubles, but I followed [this](https://github.com/Bodmer/TFT_eSPI/issues/118) and had it working. The code I am using is the example code (Keypad_240x320). The code works perfectly. First, it runs the touch_calibrate() function, which for the most part, runs the line tft.calibrateTouch(calData, TFT_MAGENTA, TFT_BLACK, 15). This function prints 4 arrows in the four corners of the screen, and after touching these arrows, displays the keypad. 

I've seen that the code also has a tft.setTouch(calData); in the if condition, in case the calData is already available in the SD card. In case it is, the tft.calibrateTouch(calData, TFT_MAGENTA, TFT_BLACK, 15) is skipped.

I am creating a device for a user and I do not want the user to go through the calibration every time the device is powered on. I made modification in the code to print out calibration values from tft.calibrateTouch(calData, TFT_MAGENTA, TFT_BLACK, 15) function, and then wrote a new function that had the following contents:

`void touch_calibrate1(){
    uint16_t calData[5] = {322,322,322,322};
//  tft.calibrateTouch(calData, TFT_MAGENTA, TFT_BLACK, 15);
    tft.setTouch(calData);
    Serial.println("printe");
  }
`
The values 322 are ones printed out by the tft.calibrateTouch(calData, TFT_MAGENTA, TFT_BLACK, 15) function. However, when the void touch_calibrate1() gets executed and the keypad is displayed, the touch isn't registered. I'm unable to press any button. I'm sure I'm making some dumb mistake. Is there something that I'm missing out?

Bodmer, 2019-10-14:
Make sure you calibrate the screen in the correct screen rotation before using the calibration values.

At some point I will update the example to make it consistent with the updates made to the library.

UmarDastgir, 2019-10-14:
Okay. I am quite sure I am calibrating the screen in the correct screen rotation. Its the same as the keypad. I have tried doing that quite a few times. Should the touch_calibrate1() function I have written above work as expected?
P.S. I just tried the procedure again and got calibration values of {402,402,402,402}. I put these values in the touch_calibrate1() function again and it still seems to have the same problem. I know that the connections are okay because the default keypad example works perfectly.

Bodmer, 2019-10-14:
Run the [diagnostic sketch](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Test%20and%20diagnostics/Test_Touch_Controller) to make sure the touch controller is giving sensible values when the screen is touched. The values should change dramatically when the screen is touched.

Run the [Touch_calibrate sketch](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Generic/Touch_calibrate) with line 23 set to rotation 0.

The output should look something like this:
```
// Use this calibration code in setup():
  uint16_t calData[5] = { 331, 3397, 248, 3453, 2 };
  tft.setTouch(calData);
```
In the Keypad_240x320 example delete line 89 (i.e. touch_calibrate();) and replace with above lines from the calibration sketch. That should work.

UmarDastgir, 2019-10-15:
Hey Bodmer! So I followed the above and it worked like a charm! I wasn't getting correct calibration values before.
I want to thank you for everything you are doing. The library is perfect! Please let me know if I can donate! I think you are doing an amazing job. You may close the issue.

closed 2019-10-27 by Bodmer

UmarDastgir, 2019-10-28:
Thank for all your help Bodmer

summerfind, 2023-06-19:
Hi Bodmer,
None of my few pieces ILI9341 & ST7789 displaies got output by diagnostic sketch above, what can be?
all of them fake?