Bodmer/TFT_eSPI #442: ILI9331/35 support

monte-monte, 2019-10-14:
I want to add support for ILI9331 based display. I have all needed code, the only thing is that it accepts 16bit commands and parameters. With params it's simply send first half and then the second. But i can't do the same trick with 16bit command, because it will also trip RS and CS lines. Or can I?
[MCUFRIEND](http://github.com/prenticedavid/MCUFRIEND_kbv/)  library that has support for ILI9335 displays has it's own writeCommand function that accepts 16bit registers. I can do the same thing, but I want to know how better integrate this with your code, so it could be pushed upstream eventually.

Bodmer, 2019-10-14:
Unfortunately the ILI9331 requires the RS and RW bits to be included in the first byte and this library does not support that.  Significant changes to the C code would be needed. The simplest option if you wish to use TFT_eSPI on your project is to purchase a display that is already supported.

closed 2019-10-14 by Bodmer

monte-monte, 2019-10-14:
Well, I think you mean SPI protocol, but I wanted to add 8080 support for this chip. I don't see why it can't be done, if it's already done in other libraries. Unfortunately, I can't buy another display for my project, because I already have round displays with ILI9331 and parallel interface. I for sure will make them work, but I thought you would be interested in widening the driver base of your library. Oh well..

Bodmer, 2019-10-14:
Restricting the support to 8 bit parallel and the ESP32 makes the adaption much simpler.

To send 16 bit commands I suspect the only change needed (apart from adding the driver files) would be to[ change this line](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.h#L298) to send 16 bits, e.g. perhaps:

```
#define tft_Write_8(C)  GPIO.out_w1tc = clr_mask; GPIO.out_w1ts = set_mask((uint8_t)0); WR_H; \
                                   GPIO.out_w1tc = clr_mask; GPIO.out_w1ts = set_mask((uint8_t)C); WR_H
```

I do not have an ILI9331 display so am unable to help further.