Bodmer/TFT_eSPI #444: Add support for choosing BGR or RGB order on ILI9341

hodefoting, 2019-10-15:
I am using an ILI9341 display, and with other code I have been using with it the display gets initialized in RGB order, but TFT_eSPI only supports BGR order for this display. The attached patch makes the ILI9341 driver respect the already existing TFT_RGB_ORDER when initializing / setting orientation for this driver.

[0001-respect-TFT_RGB_ORDER_-defines-for-ILI9341.patch.txt](https://github.com/Bodmer/TFT_eSPI/files/3729364/0001-respect-TFT_RGB_ORDER_-defines-for-ILI9341.patch.txt)

Bodmer, 2019-10-18:
Thanks, I have updated my local copy and will update Github when the next release is ready.

closed 2019-10-18 by Bodmer

Bodmer, 2019-10-21:
Github copy has been updated.

hodefoting, 2019-10-21:
The display I have also seem to deviate from standards in a different way, it is controlled by ILI9341 commands, but is only 128x160, and the way I initialize TFT_eSPI in arduino currently is like so:

 TFT_eSPI tft = TFT_eSPI(130, 160);  // the 130 seems to help with some border glitches
 uint16_t fb[160 * 128];
void setup() {
  tft.init();

The ILI9341 implementation did not seem to like me defining the width and height, but passing them in the constructor does seem to work - but makes sketches need custom configuration in code rather than being able to pick up bits from user-setup.h  not a big issue; but yet another type of variation you might want to know of.

( you might also enjoy haveing a look at what I am doing on top of TFT_eSPI here: https://pippin.gimp.org/ctx/ )

Bodmer, 2019-10-21:
I suspect the display is not ILI9341 based but that driver just happens to work. You may have an ST7735 based display. These often have 1 or 2 pixel offsets and the library caters for this.

Post a link to the display vendors page and I will look and see if the driver is specified.

hodefoting, 2019-10-21:
Thank you, - you're right. It is the 1.8inch version of this https://www.aliexpress.com/item/32847643640.html - though scouring github I found no code initializing a 128x160 framebuffer and started guessing based on pin numbers gleaned from the ali-express documentation. Having made it work by tweaking the working ILI9341 initialization (and configuration) I guessed it had to be one of those - not knowing that the correspondence between commandsets is big. It now works as a ST7735_GREENTAB2 and it starts experiencing problems only when SPI_FREQUENCY is over 40mhz - but seems consistently stable at 38mhz.