Bodmer/TFT_eSPI #446: Information, ST7796 parallel Arduino Shield usable with Ili9488 settings

Internetwurm, 2019-10-19:
Hello Bodmer,

still awesome job with this library.

With the setting and configuration in parallel mode with an Ili9488 Shield i am also able to use a ST7796 Shield. Both are 3.5" with same resolution. Both are mcufriend shields.
I am using Wemos Lolin32 Dev Kit as MCU. A bit uncomfortable wiring but works.
The only adjustment is to comment out the "tft.invertDisplay(true);" that is needed for the Ili9488.
So hopefully i could give you some kind of support this way. 

Thanks for the sprite thing.
Cool stuff to avoid flickering.

Could be  closed.

regards

Bodmer, 2019-10-20:
Thanks for letting me know.

Do all the rotations work OK?

Internetwurm, 2019-10-20:
[image]

Hi Bodmer,
yes, so far as i could test. 
My Display content would get scrambled on vertical orientation.
The right ( grey ) one is the ili the left ( blue ) is the st.
tft.setRotation(1); //st7796
tft.setRotation(3); // ili9488

So it seems it is 180° swapped.
I cant find any other difference.
I didn´t checked touch so far.
It should work because i considered this on wiring to future use.
But also not tested so far.

Im still using ili9488 Init.h for both.

Internetwurm, 2019-10-20:
[image]
[image]
[image]

The UTFT Demo took 735-739ms on both.
And as visible on the pictures it is only 180° swapped and needs inversion for the ili9488 on the right. 
Resolution and orientation for the ST7796 looks fine for alle variations and without any "defects" or problems.

closed 2019-10-22 by Internetwurm