Bodmer/TFT_eSPI #447: eSPI Touch is not working

Mangee36, 2019-10-20:
Hi, I am not able to get touch response from TFT display in ESP32. I am using ILI9341 with XPT2046. In library in UserSetup I have right driver, I used pins listed there for ESP32 with enabled touch_cs line and defined SPI_touch_frequency. I dont think i have bad wiring. Display works well with arduino. In example test_touch_controller i get same results. x: 0     y: 0     z: 4095

my pin connections: 

VCC(display) - VIN(STM32, 5V)
GND - GND
LED - 3V3(STM32)
DC - 2
CS - 15
RST - 4
MISO - not connected
MOSI - 23
SCLK - 18
touch_CS - 21 
other touch pins not connected

Can u help me please? 
(sorry for speaking I am not native :D )

Bodmer, 2019-10-20:
You need to connect up the touch controller pins as well as the TFT, typically:
T-DO to ESP32 MISO
T_DI to ESP32 MOSI
T_CS to pin 21 already done
T_CLK to ESP32 SCLK

Mangee36, 2019-10-20:
I tried it before and it didnt make diference. I tried it now to be sure and now it works :D . Dont know where i did mistake then before. Thank you so much. I am also glad that you made this library with touch included. Nice work. :D Have a nice day

closed 2019-10-21 by Bodmer