Bodmer/TFT_eSPI #448: 3.5 TFT LCD Shield + ESP32 -TouchScreen issue

inecoiec, 2019-10-21:
Hi, I'm so sorry if this issue has been open before, but I can not find an answer to my case.
I have a 3.5 TFT LCD Shield as this 
[image]
I want to use it with an ESP32 ussing parallel 8 bits mode connection.

I made this
- Into file _User_Setup_Select..h :_ 
comment  _#include <User_Setup.h>_
uncomment _#include <User_Setups/Setup13_ILI9481_Parallel.h>_ 

- Have connect all pins as this last file say. The display shows image without problem.

But when I want to run the _Touch_Controller_Demo.ino_ example, I get the error:
 'class TFT_eSPI' has no member named 'setTouch'

I see that I must add into setup file (in my case _User_Setups/Setup13_ILI9481_Parallel.h_ ) the _#define TOUCH_CS_ declaration with the pin attached T_CS display pin. 
The first problem is my display has not a T_CS pin :(.
I have made this declaration anyway (with any pin) , and get this error:  _Error compiling for board Node32s._

As a last commentary, I tell that If I attach this display Arduino One, can use the touch screen without problem.

Wait for your comments and sorry for my English !

Bodmer, 2019-10-21:
The library only supports touch screens based on the XPT2046 chip. Typically UNO style shields have an analogue interface. Do not define the T_CS pin for the UNO style display.

You could [try this library](https://github.com/s60sc/Adafruit_TouchScreen), you will need to add further links to the bottom of the ESP32 board as detailed in that libraries documentation.

inecoiec, 2019-10-21:
Thank you Bodmer !!. 
I will try with this library and tell you about the results ...
Regards !!!

inecoiec, 2019-10-27:
Hi Bodmer,
I have tired the library you tell me, but I can not make work it
I am ussing a NodemuCu ESP32 (can be any problem with use it ?), and have made the connections as is into library description. I have made the add connections between 33 and 34 , 15 and 35 , 32 and 36 too.
I ve selected the 0x9481 display controller.
I can change the fill colour, but I can not get any data from touchscreen. In terminal I can see the display intialization but I can not see any data from touch activity.

Is there some other thing that I can do ??
Many thanks a lot !!
Omar

Bodmer, 2019-10-28:
Are you sure the screen has touch capability. Post a picture of the screen side of your actual board, not one from the vendors web page.

inecoiec, 2019-10-28:
Hi Bodmer:
I want to tell you that I can use the touch capability when I attach the display to Arduino one.

I have confirmed that the drawing is ok. 
Is possible  that I must do any change into any library  ???

Here are the images. 
[image]
[image]

Thanks !!!!

Bodmer, 2019-10-29:
I suspect you have not wired it correctly.
See the [pictures here](https://github.com/Bodmer/TFT_eSPI/tree/master/Tools/ESP32%20UNO%20board%20mod). Notice how some pins of the TFT shield go to 2 different ESP32 pins and this is important. Take note of the [extra wiring here](https://github.com/s60sc/Adafruit_TouchScreen) too and think through what ESP32 GPIO to shield pins are being connected.

It is much easier to buy an UNO style ESP32 board.

closed 2019-11-02 by Bodmer

gvfo5, 2020-12-24:
Hi, I am noob and I bought the same display and esp 32 board as mentioned. Can any help me with the connection? I was trying to build a freedesk macro button https://www.instructables.com/A-Bluetooth-ESP32-TFT-Touch-Macro-Keypad/

sebbe2407, 2021-01-15:
Hi everyone... 

i had the same issues with this 3,5" TFT LCD and wiring it to an ESP32 and making the TouchScreen work. However i managed to find a solution to the problem. Lets start with the wiring:  

#define TFT_CS   33                 
#define TFT_DC   15  
#define TFT_RST  32
#define TFT_WR   4 
#define TFT_RD   2 
 
#define TFT_D0   12  
#define TFT_D1   13  
#define TFT_D2   26  
#define TFT_D3   25  
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14

Next lets focus on the software side. In bodmers awesome library you have to comment/uncomment the right sections. The User_Setup.h file is pretty straight forward. My display used the ILI9488 processor and is run in 8 Bit parallel mode: 
 
#define ILI9488_DRIVER             
#define TFT_PARALLEL_8_BIT

When it came to the touchscreen i faced some difficulties. Since the display with 8 bit there is no dedicated "Touch Pin" like other displays use. I decided to use a different library than @Bodmer (no front, still love you and the library <3). I used the ADAFRUIT Touchscreen library in extend: 

#include <stdint.h>
#include "TouchScreen.h"

The Touchscreen sends us four different values if wired correctly. The X-Position, Y-Postition aswell as the pressure applied (XM, YM). These values are received via the following pins: 

XP = TFT_D0
YP = TFT_CS
XM = TFT_DC
YM = TFT_D1

This did the trick for me and made the touchscreendemo example work for me. 

Maybe this helps some people ....

Have a good one ! :) @gvfo5

Bodmer, 2021-01-16:
The simplest solution for the 8 bit parallel boards is to use an UNO format ESP32 board and make the changes as detailed in the library ReadMe. If you hand wire then the wires must be kept short and also use several GND(0V) wires to improve signal integrity.

Thanks to @gvfo5 for the setup info.

valentinoeval, 2022-09-24:
Hi, TFT_Dx are absolutely mandatory ? Normaly SPI it's 4/5 pins not more

Bodmer, 2022-09-24:
These instructions are for an 8 bit parallel with resistance interface (analogue signals direct to processor, i.e. no XPT2046 SPI controller fitted).

Robymak, 2022-10-21:
Hi Bodmer,
I am not very familiar with Arduino software and English, helped by a translator (sorry).
I have a problem with my 3.5" TFT used with the ESP32.
I used an 8-bit bus which I set up with a connection as shown by Sebbe2407 to the processor with:
#define TFT_D4 17
#define TFT_D5 16
My problem is that pins 16 and 17 of the processor are used for the inputs of an optical encoder and its interrupt, which I don't know how to change.
I have tried reconfiguring and wiring them to pins 18 and 19 of the processor, but the interrupt does not work. 
What could I do to liberate pins 16 and 17?
Do you have any suggestions?
Thank you in advance, Roberto

Bodmer, 2022-10-21:
Change pins TFT_D4 and TFT_D5 to 18 and 19. Run Read_User_Setup to check the compiler is using the correct pins. The output of the sketch is to the Serial Monitor window only.

The library should not affect interrupt operation so this is not something I can help with.

Bodmer, 2022-10-21:
If you mean the XPT2046 interrupt then be aware the library does not use the interrupt to detect touches.

inti007, 2023-04-28:
hola al final lo resolvi tengo el tactil funcionando 100% , al que le interese me preguntan por aca. solucion esp32 ,ili9486, conexion parelo . tactil funciona.

pabloc79, 2023-06-21:
Buenas, como estas inti007 me podes dar una mano estoy renegando con mi esp32 y la ili9488 ili 9486

inti007, 2023-09-02:
hla recien veo msj, decime tu problema

Karismaraju, 2024-03-10:
> Hi, I'm so sorry if this issue has been open before, but I can not find an answer to my case. I have a 3.5 TFT LCD Shield as this [image] I want to use it with an ESP32 ussing parallel 8 bits mode [...]

Hey how do you interfaced esp32 with this lcd shield. Mine is not working. Is SPI is possible with SD car pins. My lcd is not working in 3.3 v only in 5v. How do you connected the pins. I want to read the output of external dac MC4725 connected with I2C connection with microcontroller. That is plotting the wave from DAC to the lcd how can i do it

MRAbbasi1, 2024-11-05:
> Hi everyone...
>
> i had the same issues with this 3,5" TFT LCD and wiring it to an ESP32 and making the TouchScreen work. However i managed to find a solution to the problem. Lets start with the [...]

[image]

Hi, I'm newbie and i mistakenly bought this ili9488 3.5" touch screen to connect to esp32,  how can i use this?