Bodmer/TFT_eSPI #449: International fonts support

karthan, 2019-10-22:
Hi Bodmer.
Nice job with your library.
I am wondering if it is possible to have international fonts support. I am interesting particularly in greek fonts since i am building an internet radio. I have created greek fonts in .bdf and  .c format but i don't know how to incorporate them.
Thank you

Bodmer, 2019-10-22:
You have 2 options, use Adafruit_GFX compatible fonts or adding "Smooth fonts".

For Adafruit format fonts see example here:
https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Generic/Local_Custom_Fonts

Adafruit style fonts can be created by following the instructions here:
https://www.youtube.com/watch?v=L8MmTISmwZ8

Note that online converters for Adafruit_GFX compatible fonts are
available but these typically only use characters in the range 32-127,
and thus do not include the accented characters. These online converters
can however still be used with this sketch but the example characters
used must be changed.

For Smooth fonts see "Smooth Font" examples, such as [this one](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Smooth%20Fonts/Print_Smooth_Font).

karthan, 2019-10-23:
Thank you for your response.
I noticed that fonts are in a .h file.
How can i find informations about the layout of this file.
So i have 2 options.
Option A
As mentioned above  i created 2 sets of greek fonts for another project. These fonts are as .bdf , .fon or .c format. A converter from these formats to .h would be the best choice.
Option B 
Create a .h file from ttf fonts using characters 32 to 126 and 901 to 954 combined.
Again thank you

Bodmer, 2019-10-27:
A description of the Adafruit font format [can be found here](https://glenviewsoftware.com/projects/products/adafonteditor/adafruit-gfx-font-format/).

closed 2019-11-02 by Bodmer