Bodmer/TFT_eSPI #451: Bitmap to c Array

beckmx, 2019-10-28:
Hello, I was using the library, and all seem to go good, however, my doubt is when trying to use an image from a header file (.h) I noticed there are some examples using:

`tft.pushImage(0, 0, 240, 320, image);`

But the image used is using 16bit images from what I see 0xFFFF, where did you converted the images?

Bodmer, 2019-10-28:
Arrays containing FLASH images can be created with [UTFT library](http://www.rinkydinkelectronics.com/library.php?id=51) tool:
(libraries\UTFT\Tools\ImageConverter565.exe)
Convert to .c format then copy into a new tab

beckmx, 2019-10-29:
well, that's wierd, I have some colors that are converted differently, for instance gray is somehow green

Bodmer, 2019-10-29:
You will need[ this line](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Generic/TFT_Flash_Bitmap/TFT_Flash_Bitmap.ino#L46) in your sketch. Try both true and false as that affects the colours.

beckmx, 2019-10-29:
awesome!

closed 2019-10-29 by beckmx

sonictl, 2022-05-16:
/*  Create C arrays from jpeg images using this online tool:
    http://tomeko.net/online_tools/file_to_hex.php?lang=en
    If needed, first resize and crop to an appropriate width and height
    to suit your display with an image editing program such as IrfanView.
    
    You can also change the image "quality" to reduce the file size.
    
    Paste the array into a new tab, top and tail the array from the
    tool to look like the one below with:
    #include <pgmspace.h>
    const uint8_t name[] PROGMEM = {
    to start and and end with:
   };
    Change the name of the array
*/

Bodmer, 2022-05-16:
To clarify, the UTFT tool is for bitmap images and generates a 16 bit uncompressed array in 565 colour format.

The tomeko online tool is for direct conversion of jpeg images to a byte array. This is a copy of the jpeg file and thus is still compressed.

laoadam, 2022-10-12:
Hi,
is there a convert tool that can convert PNG image?
Thanks