Bodmer/TFT_eSPI #452: Experiencing white screen

eqcq, 2019-10-29:
Hi Bodmer and thanks a lot for the library.

I have an esp32 30 pins and a 320x240 ili9341. I created a user setting file like the code under.
Sometimes something display, sometimes no. Wiring seems good, and sometimes when everything is displayed suddenly with no reason screen become white after a display modification (adding a word or rectangle ...). 

Have you any idea ?

Thanks for all
```
#define ILI9341_DRIVER
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   5  // Chip select control pin
#define TFT_DC    26  // Data Command control pin
#define TFT_RST   27  // Reset pin (could connect to RST pin)

//#define USE_HSPI_PORT
#define SUPPORT_TRANSACTIONS

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

// #define SPI_FREQUENCY  27000000
#define SPI_FREQUENCY  40000000
// #define SPI_FREQUENCY  80000000

#define SPI_READ_FREQUENCY  20000000

#define SPI_TOUCH_FREQUENCY  2500000

// #define SUPPORT_TRANSACTIONS `

Bodmer, 2019-10-29:
Check your connections, power the display from 5V if it is fitted with a voltage regulator. Don't for get the GND connection.

Test using an example sketch rather than something you have created to check operation.

If this advice fails provide pictures of your setup.

kitecraft, 2019-10-31:
First, yes, Bodmer, this is a great library.  I love the Sprites.  So cool.  And the library is so fast that my display doesn't even blink despite the large number of updates per second.  
However, I've been experiencing the same problem and have been unable to trace any error.  After anywhere from 10 minutes to a few hours of running, the screen suddenly just goes white.  The program is still running without problems.  I know this because of serial output as well as the touch screen still works.  Again, i know this from serial output.  I've added a filter cap to the power but that didn't help either.  
I'll try powering with 5v and report back.  Otherwise, this problem is completely baffling me.

Bodmer, 2019-11-01:
Post a link to your ESP32 board and display and I will see if I can see anything special about your setup.

eqcq, 2019-11-01:
Hi there I finally found an issue with those settings : 
```
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   5  // Chip select control pin
#define TFT_DC    16  // Data Command control pin
#define TFT_RST   -1  // Reset pin (could connect to RST pin
```)
and I've changed the  SPI_FREQUENCY to  27000000, above create some bugs.
With that it's correct and stable since 3 days

kitecraft, 2019-11-01:
Thanks Bodmer.
Finding this specific board is a bit hard it seems.  But it's this one:  https://lastminuteengineers.com/esp32-arduino-ide-tutorial/
The only links I can find for the TFT are on ebay and amazon.  https://www.ebay.ca/itm/1pc-2-8inches-TFT-LCD-Display-Touch-Panel-SPI-Serial-ILI9341-STM32-Module-Parts-/392506408063?oid=201950756171

I am running mine @ 27mhz.  I'm going to drop it to 20 and see if that helps.

eqcq, 2019-11-01:
Hi Kitecraft reset pin set to -1 and wired on 3.3v helped too in my case

eqcq, 2019-11-01:
@kitecraft is there something engraved on the back of your board ? I've two of those type  engraved DOIT esp32 devekit1 on the back (in little)

kitecraft, 2019-11-01:
@eqcq The only thing on the back of mine is 'esp32s' in ink.  

I've been trying different SPI frequencies, as well as changing the RST pin as you described.  I'm still encountering the issue.  
I'm now trying to write a test program that will reproduce the issue.

eqcq, 2019-11-01:
@kitecraft ok with this board you've to select NodeMCU ESP-32S

eqcq, 2019-11-01:
Now that everything was fine for me; I've tried to set fequency back to : SPI_FREQUENCY  40000000 and I'm experiencing same problem that before. In my case frequency seems to be te key.

kitecraft, 2019-11-01:
Ok, I seem to have a reproducible test case now.  
[TFT_Sprite_Testing.zip](https://github.com/Bodmer/TFT_eSPI/files/3798362/TFT_Sprite_Testing.zip)
My User_Setup.h is included.
I hope it helps.  
Thank you for looking into this Bodmer.  

@eqcq   Perhaps you could also try the test case and see it if reproduces the issue for you.

Bodmer, 2019-11-01:
I have run you test sketch for 6 hours and it worked fine.

Check your setup, simplify down to just the display and ESP32.

Maybe the display is overheating?

kitecraft, 2019-11-01:
Thanks for looking Bodmer.

kitecraft, 2019-11-02:
Update:  (last one from me I guess)
This whitescreen problem seems to be only on that one specific type of board.  I've reproduced the issue on 2 boards and 2 displays (and combinations thereof).  Also, previous issues that came up during development indicate that there is some odd problem there (I can go into detail if you want),

I also happen to have a couple of the NodeMCU ESP-32S boards (https://docs.zerynth.com/latest/official/board.zerynth.nodemcu_esp32/docs/index.html) and the whitescreen is not happening on these boards (using the same displays).

So, thank you for your time Bodmer. I do appreciate you paying attention to this issue.  
And again, great library.  So useful.
You can close this issue as you like.  

Thank you.

ps.  For what it's worth, I've been a software QA Analyst and tester for the last 20 years.  So, I'm not your Average Joe when It comes to testing for issues.  However I am not an experienced developer of any type so my code is probably not written good.

Bodmer, 2019-11-02:
Glad to hear you have it sorted. Your tests suggest this is a hardware related problem.

closed 2019-11-02 by Bodmer

ufuk5619, 2021-12-29:
Hi @kitecraft ,ı have same problem . I am using st7735 1.8 with stm32f100 mcu. for 10 to few hours it works without problem but later screen goes compeletly white.Even ı reset the board or cut the energy  its still white. Do you have any idea or any result what could be the reason.I think it is about hardware side but cant find any mistake.

Bodmer, 2021-12-29:
You may be running out of RAM, it is difficult to determine the exact cause without a detailed knowledge of the hardware and software you are running. Other possible reasons for spurious problems are EMC (long ignal wires, porr GND(0V connections) voltage spikes from motors and relays etc) and an inadequate power supply.

elisabethLC, 2022-03-31:
Hi, I have the same problem too, I am using ESP32 with 320x240 ILI9488. few days i work with my project TFT ILI9488 works fine but after i upload more code screen goes white but touch screen and SD card still works fine.

ufuk5619, 2022-03-31:
hi @elisabethLC  ı want to ask you something about lcd, are you using  lcd module or are you soldering lcd screen to any pcb. I am asking this because ı had same problems, now ı think ı solved my problems.The cause is soldering with solder paste cause some kind of effect with time and temperature. I am using no clean flux and soldering lcd so carefully and we didnt receive any error until now.

elisabethLC, 2022-03-31:
I am not directly soldering lcd screen to my pcb, I'm using pin header. I already double check the wiring and there is no problem. I tried to use breadboard with new wiring without soldering any pins and run simple code but it is not working either.

elisabethLC, 2022-03-31:
I have fixed my problem, I found something wrong with ESP3 maybe because I wrote wrong algorithm and SRAM overflowed. now works fine with new esp32. Thank you for answering my question.

2ant, 2022-04-21:
I may have something usefull to add, i got 3 ESP32 DEV KIT V2 from AZDELIVERY and i tested one. The transfer was ok once i used the boot PIN but the test was either partially or totally white. I unplugged it from my computer and plugged it on an usb powerbank and everything was crystal clear. I didn't know a lack of power could make the screen whiter.

ciobania, 2022-05-02:
I have the same issue with screen going white after some time, and could never figure out why. 
I use a Raspberry Pi Zero 2 W, and couple of sensors, for a "SMART" thermostat.

I'll try to up the power supply to 2A, and see if that helps my case.
Thanks for sharing your experiences.

GreaseMonkey88, 2023-05-24:
In my case, connecting the TFT_RST pin worked!

sahasradal, 2025-08-12:
add a 4.7k pullup to the cs pin of aliexpress st7735 lcd