Bodmer/TFT_eSPI #456: Pin mapping (esp8266)

kostuch, 2019-11-01:
Hi,

In User_Setup_Select.h you have section with pin difinitions:
```
// These are the pins for ESP8266 boards
//      Name   GPIO    NodeMCU      Function
#define PIN_D0  D0  // GPIO16       WAKE
.
.
.
```

But there is no definition for D0-D8 pins and compilation fails.
To compile I have to add before your definitions something like:
`#define D0 16`

This is strange, as for other pins you use just gpio numbers:
```
#define PIN_D9   3  //              RXD0
#define PIN_D10  1  //              TXD0
```

Is it intentional?

rumoldus, 2019-11-01:
Same problem with ESP32 with multiple pin defines that start with D.

Bodmer, 2019-11-01:
Ia m guessing you are not using the Arduino IDE?

rumoldus, 2019-11-02:
I am. 1.8.10. Sorry for not mentioning that.

Marcel Payens

kostuch, 2019-11-02:
I use AtmelStudio with VisualMicro plugin - no Arduino IDE. Suspect, that Visual Studio with VisualMicro has same issue.
I wonder if these multiple defines PIN_Dx -> Dx -> GPIOx are useful for anything?

Bodmer, 2019-11-02:
@kostuch This situation came about because for the Arduino IDE pin references D0 - D8 they are mapped to ESP8266 GPIO pins, in early ESP8266 board packages this mapping was inconsistent hence I created the PIN)Dx definitions to guarantee a fixed mapping.  Most ESP8266 boards now have consistent pin mappings, but D9 and above are not defined.
Recently I adapted the pin list to use D0 etc, this is to address issues with a small number of boards that do not map D0 to GPIO 16. Unfortunately I see there is a consequence for users who do not use the Arduino IDE.

So the best option for you is to comment out those PIN_Dx  #defines.

Bodmer, 2019-11-02:
@rumoldus It sounds like you have a setup issue. With the ESP32 you should always use GPIO numbers.

If that is not the problem then please raise a new issue.

closed 2019-11-02 by Bodmer

kostuch, 2019-11-02:
Thank you for the clarification. I've never used arduino IDE...
It's enough to use GPIO numbers (eg 16 instead of PIN_D0) in custom user setup file.
Maybe it's worth to add a comment to a file like this:

```
// For NodeMCU - use pin numbers in the form PIN_Dx where Dx is the NodeMCU pin designation
//#define TFT_DC   PIN_D0  // Data Command control pin
#define TFT_DC   16  // Data Command control pin (---no Arduino IDE, so no mapping for D0-D8 pins---)
```