Bodmer/TFT_eSPI #458: Question: Class Instantiation

hasenradball, 2019-11-03:
Hi I have a short question.

By invoking the class via `TFT_eSpi tft = TFT_eSPI()` is there the copy constructor mechanism used?

Ist it possible to call or set directly a specific rotation?
Like this:

`TFT_eSPI tft = TFT_eSPI() :: setRotation(3)`

Thanks for supporting.

Bodmer, 2019-11-03:
Instantiating the class only sets up the basic software environment. At that time the display is not initialised, so no display access functions will work until tft.begin() is called.

closed 2019-11-03 by Bodmer

hasenradball, 2019-11-03:
So you mean, moving the position definitions to the Haeder file is not possible?
Or is it possible to move the tft.init()` and `tft.setRotatio()` also to the instansiation part in the header file? 

[image]