Bodmer/TFT_eSPI #461: Pin mapping (ESP32)

rumoldus, 2019-11-03:
Hi,
With the new version 1.4.18 I am getting compile errors about pins D4, D8 not being defined.
Hardware is ESP32 Pico with a TFT7789, 240x240, No CS, 
I am using Arduino 1.18.10. I was succesfully using a slightly older version of the TFT library, maybe 1.4.8 or 1.4.11. (It had the sprite with rotation, which is terrific, already included). I decided to upgrade to 1.4.18 for no special reason other than to get rid of the warning message at the bottom of my Arduino IDE window. I never got the 1.4.18 to work, but partly due to my own actions - I never realized the old library folder with customized usersetup.h would be deleted without warning upon library upgrade. I had specific settings in there that are lost now. I got the library to compile by manually editing user setup 24 .h file to use GPIO nrs instead of D4, D8 etc. 

Now I am into even bigger trouble, I tried to go back to tft library 1.4.8 and Arduino is giving a panic error. (This is only for projects with the TFT library included) The  panic error should not be the issue for this entry. I will now first work on the Arduino IDE to compile properly, then I will get back here, but until that moment I can't test.

Bodmer, 2019-11-03:
I have released version 1.4.20 that should fix the Dx errors. This change was to try to resolve issues with old ESP8266 boards, but has caused isses with the ESP32. That is fixed now.

I have tested it with a ST7789 and it runs OK. Setupp 24 is a good starting point.

rumoldus, 2019-11-04:
You're a champ. I will test tonight. Very, very impressed by al the work you put into this.

rumoldus, 2019-11-04:
Hi Bodmer,
almost there. I have my graphics back, including rotated sprites. Only problem is, there is an offset in height. The bottom of the display is filled with colored noise and the top is shifted out of sight.
I hope you can help me with just a single statement. Height and width are both correctly set to 240.

closed 2019-11-04 by rumoldus