Bodmer/TFT_eSPI #462: small Graphics issue when Showing Time

hasenradball, 2019-11-03:
Hi,
when showing time with the following code:
```
  tft.setFreeFont(FSS12);
  tft.setTextColor(color0, color1);
  tft.drawString(timestr, time_in.x, time_in.y, GFXFF);
```

I get the following issue when the second switches from 00 -> 01 (see pic).
How can I solve this?

[image]

Thanks for your support!

Bodmer, 2019-11-03:
This occurs because proportion font strings have different lengths. Typically this can be handled by padding with spaces or use the built in setTextPadding() function, [see example](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/320%20x%20240/Free_Font_Demo/Free_Font_Demo.ino#L132).

Also consider setting the datum using [setTextDatum().](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/320%20x%20240/Free_Font_Demo/Free_Font_Demo.ino#L180)

closed 2019-11-03 by Bodmer

hasenradball, 2019-11-05:
Is there s possibility to clear a field before drawing a float value like drawfloat(20.5, 1,.... and later float(5.4, 1, ....
Then the display shows 5.45 then.
Or has this also to be done with a padding setting?

Bodmer, 2019-11-05:
Do it with padding, this is a modified version of an example that will run on a 320x240 display. Read all the comments.
[TFT_Padding_demo_320.zip](https://github.com/Bodmer/TFT_eSPI/files/3809488/TFT_Padding_demo_320.zip)

hasenradball, 2019-11-05:
Thank you very much for This demo. :-)
This is a perfect learning sketch!

hasenradball, 2019-11-05:
What exactely does the TextDatum?
This is not really clear to me.
It does not set the floating point decimal point, right?

hasenradball, 2019-11-06:
To keep the decimal point always at the same possition (1.4 -> 99.4) has nothing to do with setTextDatum, right?

Bodmer, 2019-11-06:
Correct, however the datum can be used with [the right side position function](https://github.com/Bodmer/DarkSkyWeather/blob/master/examples/TFT_eSPI_weather/TFT_eSPI_weather.ino#L557-L568) to position a float type with a fixed number of decimal places.

hasenradball, 2019-11-06:
Ist it correct, that the setTextDatum changes the referencePoint of the TFT Display?
Does it have an influence of the axes directions, then?

Bodmer, 2019-11-06:
A block of text is plotted relative to a point x,y, [see here](https://learn.adafruit.com/adafruit-gfx-graphics-library/graphics-primitives).

TFT_eSPI allows you to change that point to others, say bottom right with setTextDatum.

hasenradball, 2019-11-06:
Ok thanks the setTextDatum is to makes it easy for programmers to handle the positions.

Thanks

hasenradball, 2019-11-06:
After looking much lines of code is my understanding right, that with the function drawfloat() it is not possible to align the decimal point.
I musst use a string for the value, right?

hasenradball, 2019-11-11:
One Question I have actually.
Can your lib also used with an Arduino nano?