Bodmer/TFT_eSPI #463: ESP8266 smooth fonts and BME280

Jxid, 2019-11-06:
I tried the smooth fonts , creating fonts via prossesing.
When i load the fonts i like and print a text everything works fine.
But when i try to print a value from BME280 sensor the board restarting all the time.
I use an ESP8266 board a BME280 sensor and a TFT ST7789 color screen.

Bodmer, 2019-11-06:
Post the section of the sketch that is causing the problem or the whole sketch if you are not sure.

Jxid, 2019-11-06:
It is a very simple project just to start.
`#define ATHBI36 "Athelas-BoldItalic-36"

#include <FS.h>
#include <SPI.h>
#include <TFT_eSPI.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_BME280.h>
#include <Wire.h>

TFT_eSPI tft = TFT_eSPI();
Adafruit_BME280 bme;

void setup(void) 
{
  Serial.begin(250000);
  tft.begin();
  tft.setRotation(0);

  if (!SPIFFS.begin()) {
    Serial.println("SPIFFS initialisation failed!");
    while (1) yield();
  }
  Serial.println("\r\nSPIFFS available!");

  bool font_missing = false;
  
  if (SPIFFS.exists("/Athelas-BoldItalic-36.vlw")    == false) font_missing = true;
  
  if (font_missing)
  {
    Serial.println("\r\nFont missing in SPIFFS, did you upload it?");
    while(1) yield();
  }
  else Serial.println("\r\nFonts found OK.");

  tft.fillScreen(TFT_BLACK);        
}

void loop() 
{
  tft.setTextColor(TFT_YELLOW, TFT_BLACK);
  tft.loadFont(ATHBI36);
  tft.print("Θερμ. :");tft.println(bme.readTemperature(),1);
  tft.print("Υγρ.  :");tft.println(bme.readHumidity(),0);
  tft.print("Πιεσ. :");tft.println(bme.readPressure(),1); 

  delay(5000);   
}`
I have create the font "Athelas-BoldItalic-36.vlw" via Prossecing.
I have upload it to ESP8266 via the tool in Arduino IDE.
I used the Greek and English characters of the font because i need greek characters to print.
If i remove the sensors values from printing in tft the display works great.
It displays the words i need in Greek.

In Serial monitor i get these:
`SPIFFS available!

Fonts found OK.

Exception (28):
epc1=0x40203d64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffdc0 end: 3fffffc0 offset: 01a0
3fffff60:  3ffe860e 00000000 3fffff90 40204d78  
3fffff70:  3ffe85f8 3ffee318 3ffee318 40203e50  
3fffff80:  3fffdad0 00000000 3ffee360 40201128  
3fffff90:  00000000 00000000 fffee584 402010be  
3fffffa0:  3fffdad0 00000000 3ffee5ac 40205248  
3fffffb0:  feefeffe feefeffe 3ffe8508 40100521  
<<<stack<<<`
Any help would be apreciated.
Thank you,
Yannis

Bodmer, 2019-11-06:
It looks like you have not initialised the sensor library.  [Add the code here]([url](https://github.com/adafruit/Adafruit_BME280_Library/blob/master/examples/bme280test/bme280test.ino#L42-L55)) and see what is reported.

Jxid, 2019-11-06:
Thank Bodmer.
It was my mistake.
I haven't see that i had no initilization of sensor.
I am new to programming and i miss things.
Thanks again for your fast answer and for your help.
Yannis.

Bodmer, 2019-11-06:
So it worked then?

closed 2019-11-06 by Jxid