Bodmer/TFT_eSPI #464: R61509V Chip TFT Display Driver

Jxid, 2019-11-06:
I like your library very much.
Is there any chance to support the R61509V controler?
I have a 3.0" tft touch display with R61509V chip driver and i would like to use it with your library.
Thank you,
Yannis

Bodmer, 2019-11-07:
I do not have one of those displays so testing would be a problem. I think that driver needs 16 bit commands so that complicates matters a bit.
Where did you get the display from?

Jxid, 2019-11-07:
I had bought it from eBay.
I will post the link when I get home and I will post a photo to see how it is. It is an open-smart display.
Thanks for your consern.
Yannis

Jxid, 2019-11-07:
These are the photos of my display.

[image]
[image]
[image]
[image]

Bodmer, 2019-11-10:
OK, it is an 8 bit parallel display. Wiring these to an ESP32 is rather problematic due to the high edge speeds of the signals. I suggest you get an ESP32 UNO style board and ILI9481 board as that will be more reliable.

Jxid, 2019-11-10:
OK thank you for the info.
I will try something else.

closed 2020-01-11 by Bodmer