Bodmer/TFT_eSPI #466: Use a function for background value in drawGlyph

delhoume, 2019-11-07:
I have a use case where I have a gradient as a background for the whole screen and I would like smooth fonts to display well in this case, regardless of position in screen. Currently the background is set to a single color and alphaBlended with current text pixel color. This gives bad results even within a single glyph when bg is not uniform across the glyph.

I would like to have a callback function where the background is computed for each global x, y.
This way I could have correct anti-aliasing on my text on any algorithmically computable background (also on background images for example)

that would be something like

`typedef uint16_t (GETREALBG)(uint16_t x, uint16_t y);
`
in `::drawGlyph(uint16_t index,  GETREALBG GetBg = 0) { `
line 512 `drawPixel(x + cx, y + cy, alphaBlend(pixel, fg, GetBg ? GetBg(x + cx, y + cy) : bg)));`
and in my case
```
uint16_t getRealGradientBg(uint16_t x, uint16_t y) {
 return gradient_color(x, y);
}
```

Bodmer, 2019-11-08:
This is an interesting idea.  It will speed up incorporating and testing this potential update if you post an example sketch that I can use.

delhoume, 2019-11-08:
Here is a snippet:
```
typedef struct RGB {
  uint8_t r;
  uint8_t g;
  uint8_t b;
} RGB;

uint8_t lerp(uint8_t v1, uint8_t v2, float p) {
  return v1 + (uint8_t)(p * (v2 - v1));
}

uint16_t interpColor(RGB color1, RGB color2, float p) {
   return tft.color565(lerp(color1.r, color2.r, p), lerp(color1.g, color2.g, p), lerp(color1.b, color2.b, p));             
}

// test with very different colors
RGB topColor = { 105, 155, 230 };
RGB bottomColor = { 45, 85, 205 };
 
uint16_t getBackgroundColor(uint16_t x, uint16_t y) {
   float p = y / (float)tft.height();
   return interpColor(topColor, bottomColor, p);
}

void fillGradient() {
     uint16_t w = tft.width();
    for (uint16_t y = 0; y < tft.height(); ++y) {
      uint16_t color = getBackgroundColor(0, y); // x not used here
      tft.drawFastHLine(0, y, w, color);                     
  }
}

uint16_t fg = TFT_WHITE;
uint16_t darkblue = tft.color565(65, 105, 225);

void setup() {
tft.begin();
fillGradient();
 tft.setTextColor(fg, darkblue);
drawSomeSmoothTextInVariousPositions();
}

void loop() { yield(); }
```

as will see there is no way to make the background text color work for all positions of text in screen. And even within a text, it is not always correct as each pixel is different.
In this case the function `getBackgroundColor` could be used directly as a background color callback when drawing anti-aliased text, returning the correct x and y background color for any pixel on the screen.

Bodmer, 2019-11-10:
I have added the required functions and 2 new examples. Thanks for the suggestion.

Report back if you have any issues.

Bodmer, 2019-11-10:
Seems to work:
https://i.imgur.com/YMBcPHp.png

Bodmer, 2019-11-10:
[https://i.imgur.com/YMBcPHp.png](url)

Bodmer, 2019-11-10:
[image]

delhoume, 2019-11-10:
Very nice pixel perfect result, thanks.

closed 2019-11-11 by Bodmer