Bodmer/TFT_eSPI #468: ILI9481 8bit parallel with ESP32 not working

Puebllo, 2019-11-10:
Hi Bodmer,
first of all Thank You for your work on this library, it's great, but i have some dificulties with running LCD with driver ILI9481 in parallel mode on ESP32. I'm using bare LCD screen (without touch screen) from this seller:
https://www.aliexpress.com/item/32813655933.html?spm=a2g0s.9042311.0.0.27424c4dFtvcs0

Arduino IDE version: 1.8.9
Board: ESP32 Dev Module

On very old library version - 0.16.14 LCD works so I think there is no hardware problem. Here is 
Read and User setup output of newest library:
```
TFT_eSPI ver = 1.4.20
Processor    = ESP32
Frequency    = 240 MHz
Transactions = Yes 
Interface    = Parallel 
Display driver = 9481
Display width  = 320 
Display height = 480 

TFT_CS   = D33 (GPIO 33)
TFT_DC   = D15 (GPIO 15)
TFT_RST  = D32 (GPIO 32)

TFT_WR   = D4 (GPIO 4)
TFT_RD   = D2 (GPIO 2)

TFT_D0   = D12 (GPIO 12)
TFT_D1   = D13 (GPIO 13)
TFT_D2   = D26 (GPIO 26)
TFT_D3   = D25 (GPIO 25)
TFT_D4   = D17 (GPIO 17)
TFT_D5   = D16 (GPIO 16)
TFT_D6   = D27 (GPIO 27)
TFT_D7   = D14 (GPIO 14)

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

```
Here is list what I've tried:
*New libraries installed via Library manager
*Tested on libraries from 1.4.10 to 1.4.20
*LCD was tested with my own sketch implementing TFT_eSPI and example: 480x320 > TFT_Print_Test  (both working on old library) 
*I was also trying to use HX8357D_DRIVER with no luck
*Predefined Setup - Setup13_ILI9481_Parallel.h also not working
*Pin mapping from old library isn't working in new library and vice versa
*Reset pin set to -1 and connected to 3.3V

Ultimately i want to drive this lcd with SPI.

Here is link to my User_Setup.h and configured library with which LCD works in 8bit parallel mode:
[Old library](https://drive.google.com/file/d/1J0vZHaKCshr9QPUp6FOfPrst7lY0vpFk/view?usp=sharing)
[User_Setup.h](https://drive.google.com/file/d/1fzkCPLtVkO8397JYXOH7-0xjU4ml4jVy/view?usp=sharing)

Hope, the bug is easy to fix :)

Bodmer, 2019-11-11:
Which ESP32 board package version are you using (check via IDE Boards Manager)?

Puebllo, 2019-11-11:
Newest version - 1.0.4

Bodmer, 2019-11-11:
OK, I have tested the library with three displays and they all work OK with the setup file you have used. These displays are all UNO style shields and connect to an UNO style ESP32 board. The boards do use  HC245 CMOS buffers between the ESP32 and the TFT.
The performance of the library has been improved and this may be an issue if you have "wired" the display to the screen. The ESP32 outputs very fast edges and this can cause ringing and cross-talk on the connections if they do not pass over a ground plane. The WR line is particularly critical so try adding a 22 Ohm resistor in series placed at the ESP32 end of the connection if you are not using buffers. Adding extra GND lines between the TFT and ESP32 may help too.

Can you post a picture of your setup please so I can see how the display is connected to the screen.

Puebllo, 2019-11-11:
I've added 22ohm resistor like you said but nothing has changed. But I also have good news (probably), when I've touched the wires, moved them around, the LCD displayed some garbage - white background with alternating black lines or white backround and some noise all over the screen

[image]
[image]

Breadboard is used to connect power, IMx pins and unused pins accordingly to datasheet. Data pins from ESP32 are going straight to FPC breakout board

Right now I'm also waiting on PCB to this LCD which utilize HC245 buffers so LCD could be 3,3V / 5V compatible

Bodmer, 2019-11-11:
Hi, I connected up a display like yours (ILI9481)  and it worked fine with Setup 13. You may find there is an intermittent connection somewhere. Colours were inverted though so I added this line in the setup():
tft.invertDisplay( true );

[image]
[image]

Puebllo, 2019-11-11:
Thank You once again for your help. I think i found the problem. The ESP32 itself cannot boot properly:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK
>ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
```

I think when I solve this problem we might return to this issue.

closed 2019-11-11 by Bodmer

dexter323i, 2021-07-11:
Hi @Bodmer !
First thank you so much for your TFT_eSPI, it is a huge help for my project!

Are you sure your image is correct about the pinout? I tried to follow that, and I get a white screen. :(
-TFT pin 10 (RS/SPI_SCL) goes to GPIO 2
-TFT pin 12 (RD) also goes to GPIO 2
Both goes to pin 2 on ESP32?

Here are the pinouts from the setup file:
#define TFT_CS   33
#define TFT_DC   15
#define TFT_RST  32
#define TFT_WR    4
#define TFT_RD    2

I can't see GPIO 15 on your picture. Where do you connect this?

dexter323i, 2021-07-15:
For future visitors who arrive here, with a white screen problem, and who also has WROVER version of ESP32:
It does not have IO16 and IO17 pins! That is why Bodmers pinout not works with WROVERs. This is what works for me instead:

#define ILI9488_DRIVER
#define ESP32_PARALLEL

#define TFT_CS  4  
#define TFT_DC  13 // called RS/SPI_SCL on Display side
#define TFT_RST 32
#define TFT_WR  14
#define TFT_RD  -1  // Not connected

#define TFT_D0   18
#define TFT_D1   19
#define TFT_D2   21
#define TFT_D3   22
#define TFT_D4   23
#define TFT_D5   25
#define TFT_D6   26
#define TFT_D7   27

alvarotorijano, 2021-11-01:
Which interface is faster 8bit or SPI on an esp8266 or esp32? (sorry for asking here but I did not found any discussion about this)